wasp-os/wasp-os #1: ModuleNotFoundError: No module named 'click'

tgwaste, 2020-02-02:
:(

CC build-pinetime-s132/pins_gen.c
CC build-pinetime-s132/frozen_content.c
LINK build-pinetime-s132/firmware.elf
   text    data     bss     dec     hex filename
 178760     152    1772  180684   2c1cc build-pinetime-s132/firmware.elf
arm-none-eabi-objcopy -O binary build-pinetime-s132/firmware.elf build-pinetime-s132/firmware.bin
arm-none-eabi-objcopy -O ihex build-pinetime-s132/firmware.elf build-pinetime-s132/firmware.hex
make[1]: Leaving directory '/root/wasp-os/micropython/ports/nrf'
python3 -m nordicsemi dfu genpkg \
        --dev-type 0x0052 \
        --application micropython/ports/nrf/build-pinetime-s132/firmware.hex \
        micropython.zip
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/root/wasp-os/tools/nrfutil/nordicsemi/__main__.py", line 32, in <module>
    import click
ModuleNotFoundError: No module named 'click'
make: *** [Makefile:34: micropython] Error 1

tgwaste, 2020-02-03:
After doing:
`pip3 install click serial`

Im getting:

```
CC build-pinetime-s132/frozen_content.c
make[1]: Leaving directory '/root/wasp-os/bootloader'
python3 -m nordicsemi dfu genpkg \
        --bootloader bootloader/_build-pinetime_nrf52832/pinetime_nrf52832_bootloader-*-nosd.hex \
        --softdevice bootloader/lib/softdevice/s132_nrf52_6.1.1/s132_nrf52_6.1.1_softdevice.hex \
        bootloader.zip
Failed to import ecdsa, cannot do signing
LINK build-pinetime-s132/firmware.elf
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/root/wasp-os/tools/nrfutil/nordicsemi/__main__.py", line 37, in <module>
    from nordicsemi.dfu.dfu_transport_serial import DfuTransportSerial
  File "/root/wasp-os/tools/nrfutil/nordicsemi/dfu/dfu_transport_serial.py", line 37, in <module>
    from serial import Serial
ImportError: cannot import name 'Serial' from 'serial' (/usr/local/lib/python3.7/dist-packages/serial/__init__.py)
make: *** [Makefile:18: bootloader] Error 1
make: *** Waiting for unfinished jobs....
   text    data     bss     dec     hex filename
 178760     152    1772  180684   2c1cc build-pinetime-s132/firmware.elf
arm-none-eabi-objcopy -O binary build-pinetime-s132/firmware.elf build-pinetime-s132/firmware.bin
arm-none-eabi-objcopy -O ihex build-pinetime-s132/firmware.elf build-pinetime-s132/firmware.hex
make[1]: Leaving directory '/root/wasp-os/micropython/ports/nrf'
python3 -m nordicsemi dfu genpkg \
        --dev-type 0x0052 \
        --application micropython/ports/nrf/build-pinetime-s132/firmware.hex \
        micropython.zip
Failed to import ecdsa, cannot do signing
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/root/wasp-os/tools/nrfutil/nordicsemi/__main__.py", line 37, in <module>
    from nordicsemi.dfu.dfu_transport_serial import DfuTransportSerial
  File "/root/wasp-os/tools/nrfutil/nordicsemi/dfu/dfu_transport_serial.py", line 37, in <module>
    from serial import Serial
ImportError: cannot import name 'Serial' from 'serial' (/usr/local/lib/python3.7/dist-packages/serial/__init__.py)
make: *** [Makefile:34: micropython] Error 1

```

tgwaste, 2020-02-03:
Looks like this is the answer.  May want to include it in instructions:

`pip3 install click serial pyserial ecdsa`

daniel-thompson, 2020-02-03:
Thanks Ted.

I plan to look at improving the instructions once I have most of the M1 features on place. See [TODO](https://github.com/daniel-thompson/wasp-os/blob/master/TODO.md) for the roadmap. Hopefully that well be pretty soon now... I made some really good progress over the weekend!

jahoops, 2020-02-03:
I bought two Pinetime watches so that I could do some fun tandem development with my daughter (recent college grad, some JavaScript experience).  I am a software developer who likes to tinker with electronics and build things (cheap sous vide, yogurt maker, dough proofer, houseplant monitor) but I am only hobbyist level in electronics and bit banging. 

After unlocking the Pinetime with a RPi and compiling wasp-os on Linux, I did manage to (on Windows) copy the bootloader.hex using an ST-Link and have the small pine cone logo on the watch.  The (Android) Nordic connect program links to the watch, but I keep getting a few errors and I don't see the larger multi colored pine cone, so I assume it isn't working yet, but I'm close.

Anyway, if you like I could work on documenting a reliable way to set up the Pinetime using the most common tools, like Windows and VSCode.  I do think many people would derive a lot of satisfaction from programming their own smart watch!

Sorry to post this where it obviously doesn't belong, but I didn't see any other way to get a message across.  Please delete this if you like and email me at jwhooper@gmail.com if you would like me to work on this most egalitarian approach.

commit 5d2ca4a8 (2020-05-20): README: Better document how to meet the dependancies

Fixes: #1
Fixes: #3
closed 2020-05-20 by daniel-thompson