wasp-os/wasp-os #2: [?] Can't fine stdio.h and others

MarcusE1W, 2020-02-05:
Hi.
I am woefully aware that this will end up to be a very stupid question, but in the moment I am stuck.

I try to compile the firmware with `make -j `nproc` BOARD=pinetime all` and get the following error message:
```
GEN build-pinetime-s132/genhdr/qstr.i.last
In file included from ../../py/mpstate.h:29:0,
                 from ../../py/mpstate.c:27:
/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h:9:26: fatal error: stdint.h: N
o such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
...
```
(and a lot more, all about std. includes not found)

I am not really familiar with C but on my search for a solution I just compiled a simple "hello world" progamme and it seems that works fine, the includes get found. Now the target is obviously a different one for the firmware. Do I have to set something up so that gcc finds the right stdint.h for the Pinetime?

when looking for stdint.h I get these folders:
```
~/prog/pine/wasp-os$ find /usr -name stdint.h

/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h
/usr/lib/gcc/arm-linux-gnueabihf/7/include/stdint.h
/usr/include/stdint.h
/usr/include/c++/7/tr1/stdint.h
```
I feel I am missing something stupid. Any thoughts would be appreciated.

jahoops, 2020-02-06:
You have to set the paths:
export PATH=$PATH:/place/with/the/file

daniel-thompson, 2020-02-06:
I am using the GNU-RM toolchain (9-2019-q4) for all my builds.

You will certainly have problems with the 6.3.1 compile you have picked up from your distro (they are not insurmountable but there will be problems, especially with link-time optimization).

closed 2020-04-07 by daniel-thompson