wasp-os/wasp-os #3: [Info] List of dependencies

MarcusE1W, 2020-02-05:
Hi.

As I set up this project from scratch I thought I share the list of all dependencies that I have found so far.

```
sudo apt install gcc-arm-none-eabi
sudo apt install python3
sudo apt install python3-pip

pip3 install --upgrade setuptools
pip3 install wheel

pip3 install click serial pyserial ecdsa
```

This is just for info and can be closed.

tht-jxny, 2020-05-05:
In my case in order to run the simulator I had to install `pysdl2`:
```
pip3 install -U pysdl2
```

jlukanc1, 2020-05-17:
Pysdl2 seems to not work properly on Chrome OS Linux beta on an ARM CPU. Should be noted that by default, pip installs to python 2.7. Pip3 did not fix the issue for me.

daniel-thompson, 2020-05-18:
pysdl2 has requirements that cannot be met using pip3 (see https://pypi.org/project/PySDL2/ ). I'd normally recommend installing pysdl2 from you package manager rather than pip to ensure the other requirements are met (`apt install python3-sdl2` or the equivalent for your distro).

jlukanc1, 2020-05-19:
"unzip" is required to make softdevice
Manjaro does not include it by default, not sure about pinebook Manjaro.

commit 5d2ca4a8 (2020-05-20): README: Better document how to meet the dependancies

Fixes: #1
Fixes: #3
closed 2020-05-20 by daniel-thompson