wasp-os/wasp-os #4: Link failure

mulm, 2020-02-06:
Fetched wasp-os today and followed instructions from README.md .  Success until very last step of `make -j `nproc` BOARD=pinetime all`.  Failed to link:
```
LINK build-pinetime-s132/firmware.elf
build-pinetime-s132/frozen_content.c:165:26: error: size of 'mp_qstr_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t mp_qstr_const_pool;
                          ^
../../py/qstr.c:103:19: note: 'mp_qstr_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_const_pool = {
                   ^
../../py/qstr.c:118:26: error: size of 'mp_qstr_frozen_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t MICROPY_QSTR_EXTRA_POOL;
                          ^
build-pinetime-s132/frozen_content.c:166:19: note: 'mp_qstr_frozen_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_frozen_const_pool = {
                   ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Makefile:428: recipe for target 'build-pinetime-s132/firmware.elf' failed
make[1]: *** [build-pinetime-s132/firmware.elf] Error 1
make[1]: Leaving directory '/media/will/NixSSD/src/pine64/pinetime/wasp-os/micropython/ports/nrf'
Makefile:31: recipe for target 'micropython' failed
make: *** [micropython] Error 2
```

daniel-thompson, 2020-02-07:
I think this is caused by your arm-none-eabi toolset. I suspect you have picked up a relatively old version included in your distro? Lot's of people seem to be having problems due to that so I have added a couple of notes to the README describing the toolset I am using.

I suspect you could also work around by adding LTO=0 to the make invocation (this approach isn't really tested much and makes the binary a good deal bigger so I'd recommend getting a better toolset instead).

mulm, 2020-02-08:
Yep, that was it!  Dev machine is Ubuntu 18.04.4 LTS and system package for gcc-arm-none-eabi was 6.3.1 .  Pulled down the recommended one mentioned in [README.md](../blob/master/README.md#building-from-a-git-clone) (from [developer.arm.com](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) ) -- and had success:
```
LINK build-pinetime-s132/firmware.elf
   text    data     bss     dec     hex filename
 183828     152    1776  185756   2d59c build-pinetime-s132/firmware.elf
arm-none-eabi-objcopy -O binary build-pinetime-s132/firmware.elf build-pinetime-s132/firmware.bin
arm-none-eabi-objcopy -O ihex build-pinetime-s132/firmware.elf build-pinetime-s132/firmware.hex
make[1]: Leaving directory '/media/will/NixSSD/src/pine64/pinetime/wasp-os/micropython/ports/nrf'
python3 -m nordicsemi dfu genpkg \
        --dev-type 0x0052 \
        --application micropython/ports/nrf/build-pinetime-s132/firmware.hex \
        micropython.zip
Zip created at micropython.zip
```

Now, on to replacing stock firmware on my PineTime!

closed 2020-02-08 by mulm

thiswillbeyourgithub, 2022-01-17:
Hi, I had the same issue and struggled to update gcc-arm-none-eabi from 6.3.1 on ubuntu 18.04 to version 10.3 so I'm posting this here, hoping that it can help others.

Here's roughly the steps I had to go through :

* open this page and download the latest linux tarball : https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

then following this page : https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa

* sudo apt remove gcc-arm-none-eabi
* cd /usr/share
* sudo tar xvjf gcc-arm-none-eabi-YOURVERSION.bz2 -C /usr/share/
* sudo ln -s /usr/share/gcc-arm-none-eabi-YOURVERSION/bin/* /usr/bin/