wasp-os/wasp-os #6: Python scripts not running and bootloader not responding to button press.

salcedo, 2020-03-12:
PineTime does not reset to the watch face after DFU micropython.zip via NRFConnect Android app. Used GNU-RM as recommended in README.md.

After DFU, I can use wasptool --console to enter REPL and enter contents of main.py manually - which starts the watch. If I long-press the button to return to boot loader, it will no longer respond to a 2 second press to get into micropython. This means I must DFU to reset the watch.

wasptool --exec does not work. When using --exec, the tool completes without errors, but no code runs on the watch and a subsequent --console enters the REPL where it presents a prompt (nothing executed).

I tried a known good micropython.zip someone else built and it too exhibits the same behaviors as my builds.

salcedo, 2020-03-13:
I've done more research into this and found the culprit.

wasp/boards/pinetime/watch.py#L68-81

daniel-thompson, 2020-03-13:
This basically means the SPI NOR is not responding to the read ID command. It is not a good sign I'm afraid.

Maybe the SPI NOR is a different part number to mine (and for some reason doesn't implement read ID) equally it could be broken.

Did the original firmware show graphics correctly... if it does the flash probably isn't broken: I don't have first hand knowledge but I heard on the grapevine that the test firmware sent from the factory keeps the icons in SPI NOR.

I guess first step is to flip the back of and try to read the part number of the SPI NOR (that's what I did when I didn't get the results I expected from my watch: https://forum.pine64.org/showthread.php?tid=9092 ). In my case the marking were really faint and impossible to read with the naked eye. I ended up taking high res photos with the watch strongly cross lit (a window sill in my case ;-) ) and reading the part number of the pictures.

The SPI NOR is the second largest chip in the photo below (near the middle, 8 pins) and, as you can see, wasn't lit well enough in this photo to read the part number.

[image]

daniel-thompson, 2020-03-13:
PS If you want to at least have the watch working whilst you study your SPI NOR you can comment out everything in watch.py after "mount the flash section" (except maybe the backlight set) and add main.py to wasp/boards/pinetime/manifest.py . This will run everything from the nRF52's built in flash. The development experience won't be as nice because, instead of just tweaking main.py with an extra import, you'll have to do full OTA updates every time you want to make durable changes to the code. However it should work and you will still be able to use ./tools/wasptool --console to launch the REPL and interactively send commands to the SPI flash so you can figure out how it works!

salcedo, 2020-03-13:
The original firmware worked fully. It showed graphics and icons correctly.

Confirmed flash part is XT25F32B.

Confirmed with multimeter that its CS pin is properly mapped. In REPL, tried SPI.init() with all possible baudrate of NRF_SPI_FREQ_* to get something back from RDID. Always zeros.

It will return all zeros trying to RDSR or RDCR as well. The screen works, thus spi works. The CS mapping works. Flash chip does not respond to any commands.

Good idea on the workaround. Thank you! I'll add main.py to the manifest.py and remove the  mount flash stuff from watch.py.

I'd like to leave this issue open so others can find it if they run into this. But will follow your decision.

bergzand, 2020-03-15:
@daniel-thompson Not sure if it is related to this issue but from what I've seen is that the stock firmware image puts the SPI NOR chip in a sleep mode. Thus if a new firmware is flashed without power cycling the PineTime, the SPI flash chip is still in sleep mode and won't respond to commands. This can be fixed by either waking up the flash chip before attempting anything else with the flash or by power cycling the device after removing the stock firmware.

salcedo, 2020-03-15:
@bergzand You are correct! This was the issue. I sent 0xAB to the chip and it woke up.

```
>>> flash = FLASH(spi, (Pin('NOR_CS', Pin.OUT, value=1),))
1 chips detected. Total flash size 4MiB.
```

Thank you for solving the mystery. :+1:

closed 2020-03-15 by salcedo

daniel-thompson, 2020-03-16:
@bergzand: Thanks for the hint! By the sound of it I dodged this problem because I did a battery run-flat test to check if I had got the battery meter code about right! Naturally that also cleared the SWD debug flag and doubled the battery life too.

PR #7 (2020-04-10): wasp: pinetime: Release SPI NOR from deep power-down
Fixes #6