wasp-os/wasp-os #9: Python REPL via Nordic UART Service How-To

jmwright, 2020-03-19:
Nice work on this project. I've got everything working from on the watch side of things, it's flashed and displaying the date/time via wasp-os. My issue is with trying to access the Python REPL. In the readme it says "you can now use the Nordic UART Service", but I don't really understand what software is involved, and I don't see an obvious answer for this use case from a Google search. In one of your videos you mention that you have your laptop connected. Which OS is it, and how is that accomplished? I can connect to my PineTime using nRF UART v2.0 on Android, but it's a pain to send a Ctrl-C to stop the running program. The Bluetooth manager in Ubuntu 18.04 isn't letting me connect directly to the PineTime.

An outline of what you did to get your laptop connected would be much appreciated.

SteveAmor, 2020-03-19:
I can successfully run ```wasptool --console --verbose``` on my Raspberry Pi V3.  OS is Buster.

jmwright, 2020-03-19:
Oh wow, I was really over-complicating things. That works perfectly. Thanks @SteveAmor !

closed 2020-03-19 by jmwright