wasp-os/wasp-os #11: Add Desktop OTA-DFU Option To Readme.md

fwg-cag, 2020-04-11:
Hi!

Some people may prefer a desktop solution for OTA DFUing micropython.zip. I found that https://github.com/dingari/ota-dfu-python works well for this:
`sudo $HOME/ota-dfu-python/dfu.py -z $HOME/wasp-os/micropython.zip -a D6:62:01:XX:XX:XX`
You may want to update your docu accordingly.

BR, piner4711

PS: Great work BTW!

purlupar, 2020-04-11:
Great! In the pinetime chat, Peetz0r said that one needs to say " --legacy" for the UUID to work correctly.

https://github.com/daniel-thompson/wasp-os/issues/11

daniel-thompson, 2020-04-14:
> Some people may prefer a desktop solution for OTA DFUing micropython.zip.

Some people definitely includes me... I just haven't found anything suitable quite yet!

On the other hand, given wasp-os is a Python-based project, I'm very reluctant to recommend tools that are still based around python 2 (which is no longer supported upstream). I've done a mechanical port to python3 using 2to3 and fixed up a few obvious string versus bytes issues but it is not quite enough to get the tool running in Python 3.

I will try to take another look when I finished the current round of frenzied hacking on wasp-os itself.

daniel-thompson, 2020-04-20:
Just a quick follow up. I've managed to hack together something that runs in python3 but there's still a bit of debugging needed (in particular if you walk away from the laptop when wearing the watch then the error recovery code hits some bytes vs string issues I haven't tidied up yet).

daniel-thompson, 2020-04-26:
Fixed in the most recent round of updates..

closed 2020-04-26 by daniel-thompson