wasp-os/wasp-os #12: Documentation - Howto reboot device after loading micropython.zip?

tdaileygithub, 2020-04-16:
Great Work!  Very cool way of interacting with pinetime.

- How do you reboot the device after uploading micropython.zip?
   - I can connect to the UART service using wasptool but I can't figure out how to reset the device without going back to openocd.  I was guessing a long button press?

- What is your workflow for making changes to the python scripts on the watch?  
-- How do you edit the python files on device or do you simply reset [see above?] and upload a new zip?

tdaileygithub, 2020-04-17:
"How to develop wasp-os python applications on a Pine64 PineTime"

`https://www.youtube.com/watch?v=kf1VHj587Mc`

Exactly right!  
Maybe add link in Readme.md

closed 2020-04-17 by tdaileygithub

daniel-thompson, 2020-04-20:
Thanks for the update. Great to know that the video is useful!

PS There is already a link to it in the readme... IIRC it was added the same day I published the video.