wasp-os/wasp-os #13: Changes not showing on the Hardware

toter200, 2020-04-24:
I have been playing around with wasp-os for about a week. After the new Video i just imported the gameoflife file and registered the app. If i 'make' the simulator the app shows in the menu and if i specify it on the ring as well, When i 'make' it for the pinetime it doesn't show in either of those. This is my first programming for a micro controller so i might be missing something thats obvious.

daniel-thompson, 2020-04-25:
I've been installing the program over-the-air rather then freezing the module:
```
./tools/wasptool --exec wasp/apps/gameoflife.py --eval 'wasp.system.register(GameOfLifeApp())'
```

If you are asking why main.py doesn't get updated when you reflash wasp-os then that's a bit different. main.py is special in that it is stored in the external SPI flash rather than being frozen into the intenal flash (it is created automatically on first boot if it is not already in the SPI flash). The means you need an extra step to enable it:
```
./tools/wasptool --upload wasp/main.py
```

(you'd also have to  make sure gameoflife.py is added to the wasp/boards/pinetime/manifest.py)

toter200, 2020-04-25:
I added the file to manifest.py, i did not know i have to do that. I can change the time over the air but running 
`./tools/wasptool --exec wasp/apps/gameoflife.py --eval 'wasp.system.register(GameOfLifeApp())'
`
resolves in a pexpect EOF error. I ran this command multiple times and the error does not appear at the some percentage.

[image]

thanks for your help

toter200, 2020-04-25:
I added the gameoflife.py file to the manifest i did not know i have to do that. I can update the time over the air but trying to install the program resolves in an error. I ran the command `./tools/wasptool --exec wasp/apps/gameoflife.py --eval 'wasp.system.register(GameOfLifeApp())'
` multiple times and found that ti get two different errors that is seemingly random which one appears. Its either the pexpect end of file error

[image]

or a time error

[image]

is this a error specific to my machine?
thanks for the help

daniel-thompson, 2020-04-26:
Try fetching the latest version of wasptool. It looks like I forgot to include the changes needed to load gameoflife.py without running out of memory.

Note that you can either add gameoflife.py to the manifest (and wasptool --upload a new version of main.py that initializes is) **or** you can use (the new) wasptool to load it over the air..

toter200, 2020-05-03:
I was able to test it now, indeed the issue has be solved
Thanks

closed 2020-05-03 by toter200