wasp-os/wasp-os #14: Blank screen after wasp-os update

Izaic, 2020-05-01:
I'm just getting a blank screen after updating wasp-os today. It is backlit, but there is nothing being shown on the screen.

daniel-thompson, 2020-05-01:
Depending on how far back you are upgrading from this is probably caused by having an obsolete copy of main.py being out-of-date in your SPI flash.

You can use wasptool to update it to the latest version:

    ./tools/wasptool --upload wasp/main.py

(or if you cannot use wasptool then you can use the built-in shell module to create a new version of /flash/main.py )

Izaic, 2020-05-01:
> Depending on how far back you are upgrading from this is probably caused by having an obsolete copy of main.py in your SPI flash.
> 
> You can use wasptool to update it to the latest version:
> 
> ```
> ./tools/wasptool --upload wasp/main.py
> ```
> 
> (or if you cannot use wasptool then you can use the built-in shell module to create a new version of /flash/main.py )

This works! Thank you! I was worried there was a issue with WaspOS itself.

closed 2020-05-01 by Izaic

SteveAmor, 2020-05-09:
Just to add that after issuing the command:
```./tools/wasptool --upload wasp/main.py```
I had to DFU the firmware across again.

SteveAmor, 2020-05-23:
If you are having this problem and can only connect to the micropython repl using the ```wasptools --console``` command then here is how you can write a new main.py file to the SPI flash:
```
>>> import shell
>>> shell.cd("/flash")
# if it errors here, your flash may be in a deep sleep - follow note below
>>> shell.rm("main.py")
# ignore any error messages if you get them
>>> f=open("main.py","w")
>>> f.write("import wasp\r\n")
>>> f.write("wasp.system.run()\r\n")
>>> f.close()
>>> import machine
>>> machine.reset()
```
Note, if you can't change to directory /flash it may be in a deep sleep so follow this to wake it up https://github.com/daniel-thompson/wasp-os/issues/6#issuecomment-599206862