wasp-os/wasp-os #15: Cannot reboot PineTime

tht-jxny, 2020-05-01:
I flashed the bootloader with `openocd` and installed the `micropython.zip` using my laptop (I'm using Ubuntu 18.04).
Since yesterday it workes perfectly.
But now, I want to set the time and therefore have to connect my watch with my laptop via bluetooth.
Unfortunately, I am unable to connect the PineTime (Neither with my laptop nor with my phone). 
I wanted to reboot the watch to the state where the bootloader is visible but every time I attempt a long-press the display turns on or off but the watch never reboots. I tried to switch to a different watchface but after pressing the button for 20 seconds continuously I surrendered.

Do you have any ideas?

Btw, it would be great if you could add more information to the documentation about how to communicate with the PineTime using the terminal. Do I have to connect via bluetooth and then in the same or another terminal window simply type in the commands?
``` 
^C
import demo
demo.run()
# After watching the demo for a bit...
^C
wasp.app.draw(watch)
wasp.system.run()
```

Thanks for your answer!

tht-jxny, 2020-05-03:
Update:
I tried fully running down the battery and then kept pressing the button. Eventually, the watch switched to the bootloader and now I think I got a hang on how to enter the bootloader face. One has to long-press the button when the PineTime is in sleep mode because this will first light up the watch's display and then reboot it..
But I still cannot connect to my watch via Bluetooth - `Pairing rejected by PineDFU`.

Another question:
Could you please provide some information on how to generate Image-RLEs or in general how to use images. Is it possible to have a background image for a watchface?

daniel-thompson, 2020-05-04:
If you're having trouble with bluetooth it is usually worth trying a different device. I have a couple of laptops that just won't stay connected properly and I've had BT problems with older phones in the past.

You can generate images with tools/rle_encode.py (try --help and generally 2-bit is better than the default since you get spot colour support).

If you want a background image you would have to add the code to the watch application to do that (and wait until I get #18 fixed). Note however that photos won't be practical using RLE encoding... it is tuned to encoding spot colours (like the icons in the launcher). The compression will be very inefficient for photos (e.g. they will likely be bigger than an uncompressed images) and the decode speed will very slow too. That's not to say background photos can't be done... but you'll probably be best off copying an uncompressed image to the filesystem (tools/wasptool --upload) and then opening that and blasting it line by line to the display.

cool2man, 2020-05-05:
> Note however that photos won't be practical using RLE encoding

Yes, I agree. If I see it right, the color range restriction of the device (16 bits RGB 565) is limited using RLE 2-bit to 8 bit RGB 332. This results into only 4! availabe monochrome colors: black, dark gray, light gray, white.

daniel-thompson, 2020-05-05:
> Yes, I agree. If I see it right, the color range restriction of the device (16 bits RGB 565) is limited using RLE 2-bit to 8 bit RGB 332. This results into only 4! availabe monochrome colors: black, dark gray, light gray, white.

RLE 2-bit but supports all 256 colours from the RGB332 palette (for example the generic the app icon contains 7 colours)

I think you overlooked the palette update. When a colour that is not already in the palette is introduced then an existing colour will be evicted and a new one added. This gives access to 256 colours (and could be trivially extended to support the full 16-bit colour space by replacing the RGB332 with RGB565) but there is a two byte overhead for ever palette update (three if you adapted it for 16-bit).

This is fine when the palette gets reused several times but does not happen for photos. The image would be filled with palette updates and these updates bloat the file and make the encoding larger than the original image.

PS Additionally in the current code the technique to convert from RGB332 to RGB565 needs improvement. With a better approach we can get a stronger perceptual difference between the colours (some of the colours are a bit too similar).

tht-jxny, 2020-05-05:
As an update, I have solved any existing problems and now am in the middle of developing ( A beautiful Fibonacci Clock for my watch).  
Since our discussion mainly centers around colors and images, should we rename the issue?

After registering a new application and then starting the simulator, my program crashed with the error:
```
 File "wasp/wasp.py", line 145, in switch
    app.foreground()
TypeError: foreground() missing 1 required positional argument: 'self'
```
I then made a copy of the existing clock application simply without displaying the date to make sure, I'd not forgotten something obvious. But this try resulted in the same error.

Do you have any ideas on how to fix this?

cool2man, 2020-05-05:
> This gives access to 256 colours

Only for discussion, that's not a real problem: In the RGB 332, blue could only have one of 4 values. If you want a natural monochrome, red and green needs to have the same value. This results into the possibility of max. 4 monochrome colors (but a lot more of colored). The 256 colors of the RGB 332 supports only two grays: dark and light (and white and black).

I will try to write a PNG-8 (indexed color) solution and post it here if it works.

daniel-thompson, 2020-05-05:
> Do you have any ideas on how to fix this?

How did you register the application? I wonder if you registered the class itself rather than an instance of the class.

daniel-thompson, 2020-05-05:
> I will try to write a PNG-8 (indexed color) solution and post it here if it works.

A better conversion from 24-bit to RGB332 and then back up to RGB565 is definitely needed. The current linear RGB332 space is over simple but I'd quite like to avoid a LUT if possible.

tht-jxny, 2020-05-05:
I saved my `ClockTestApp()` class in a `clock2_test.py` file and then in main.py I added:
``` python
import wasp
from apps.clock2_test import ClockTestApp
wasp.system.register(ClockTestApp,True)
wasp.system.run()
```

I also tried adding `from apps.clock2_test import ClockTestApp` in `wasp/apps/__init__.py` and then in `wasp.py` i typed in `self.register(ClockTestApp(),True)` at the appropriate place.

The first method didn't work, the second fortunately did! Thanks a lot! 

Before I tried your Micropython approach, I visited several other repos and installed different solutions but with your repo I feel really productive. My compliments!

daniel-thompson, 2020-05-05:
Great to hear. Thanks.

BTW wasp/applications/template.py contains lots of docstrings that will eventually be expanded to become the application programmers guide. f you find anything out about the callbacks that is not answered in those docstrings raise an issue (or a pull request) because I'd like to get them as polished as possible.

cool2man, 2020-05-06:
> but you'll probably be best off copying an uncompressed image to the filesystem (tools/wasptool --upload)

--upload does not suppor binary data - do I see this right? Any other way to transfer my (binary) png image to the watch?

Thanks!

daniel-thompson, 2020-05-06:
Interesting point.

You are right... I think there would be problems with binary data. Ultimately upload is implemented by https://github.com/daniel-thompson/wasp-os/blob/dc4ea4ab6205bcdb27278367f7d5147b56cf78b8/wasp/shell.py#L64 (and wasptool then drives what the upload function expects using pexpect).

It could eventually be automated via wasptool but for now consider enabling the ubinascii module and copying a base64 encoded file to the watch. You should then be able to run a tiny bit of python on the watch itself to load the base64 file from the filesystem, decode the ascii armoured data and write it into a new file.
I think you would need to do some kind of ascii armouring either externally or internally.

cool2man, 2020-05-07:
> to run a tiny bit of python on the watch itself 

This script named testbild.py creates an binary image file. I have putted it into the wasp directory. How could I execute such a script in the simulator and how can I upload and execute such a script on real watch? Thanks!

```
#!/usr/bin/env python3

import os

# binary data, generated from testbild.png, 5388 bytes
testbild = (
    b'\x89PNG\r\n\x1a\n\x00\x00\x00\rIHDR'
    b'\x00\x00\x00\xf0\x00\x00\x00\xf0\x08\x03\x00\x00\x00\t\x8b\x19'
    ...
    b'>\xd6JDD\xff\x0f\x00&\xad_6\x10oB*'
    b'\x00\x00\x00\x00IEND\xaeB`\x82'
)
if not os.path.isfile("testbild.png"):
    (open("testbild.png", "wb")).write(testbild)

```

daniel-thompson, 2020-05-07:
Firstly, I think its really cool you are playing with this. I just needed to say that upfront because I had to follow up with quite a lot of corrections!

There's no real point in loading the above script on the simulator because the simulator does not need to simulate the SPI flash and filesystem (because python just uses the system filesystem). Running it as a regular python script will give you just as much coverage as trying to run it by hand.

Next I'm afraid I don't think the above script can be run on the device. In order to parse it the whole of that intermediate string will have to fit into RAM at the same time (I suspect twice, once as part of the source code and also after parsing) it looks like it is around 20k long.

In terms of the steps I was thinking more like this:

    # Encode the png file
    base64 my.png > my.hex
    # Upload the resulting hex file to the SPI flash (only a single
    # line will be held in RAM at one time so this easily fits)
    wasptool --upload my.hex
    wasptool -exec decodeit.py

Here decodeit.py is a file that opens both files and processes them line by line so avoid using up all the RAM. Completely untested I was imagining something like this:

    hex = open("my.hex", "r")
    bin = open("my.png", "wb")
    for ln in hex:
        data = hex.readline()
        # Completely untested but providing there are
        # an exact multiple of 4 characters per line then I think
        # we can decode this a line at a time
        bin.write(ubinascii.a2b_base64(data))
    hex.close()
    bin.close()

It would be great to extend wasptool to automate some of this but for you to get a file onto the device I think the above dance is needed.

cool2man, 2020-05-08:
Thanks! With a little change the decoder worked for me.

```
hex = open("my.hex", "r")
bin = open("my.png", "wb")
for ln in hex:
    data = ln # changed
    bin.write(ubinascii.a2b_base64(data))
hex.close()
bin.close()
```

tht-jxny, 2020-05-08:
It would be great if you could provide an example application along with a good workflow when blitting images is successful. Due to a lack of time and knowledge, I, unfortunately, cannot contribute to a solution to a degree I'd like to.

tht-jxny, 2020-05-13:
I have the next question: I'm currently working on a calculator app and therefore created several buttons as pngs. Now I want to use the `rle-encode.py` script but, unfortunately, it throws an error:
```
  File "tools/rle_encode.py", line 72
    nonlocal next_color
                      ^
SyntaxError: invalid syntax
```

My color choice cannot be the reason since I only used colors that were present in your encoded images.
Example image to reproduce the error:
[image]

daniel-thompson, 2020-05-13:
A SyntaxError is thrown when the code cannot be compiled. AFAIK nonlocal was introduced to Python in v3.0 so I would expect this to work with any python3 compiler. However I guess we could double check things... can you share some version information for your python binary?

PS Feel fee to pick any colours you like... the encoder (which requires true color PNGs to work properly anyway) should automatically convert these to the closest available.

PPS Based on the feedback from @cool2man above I plan to change the palette used for 2-bit RLE encoding shortly before the M2 release (most likely I will switch RGB332 to https://en.wikipedia.org/wiki/ANSI_escape_code#8-bit). This shouldn't matter right now but does mean that you may need to encode the images at some point in the future to keep up with the latest version of wasp-os.

tht-jxny, 2020-05-13:
I'm using python3.6. I wonder why this error is showing up now. I used this script many times in the past and everything worked fine. And I cannot remember changing anything in the file.

I now tried to encode another image - no problems.

Could you please download my uploaded file and try it out on your machine?

tht-jxny, 2020-05-13:
Oh, what a silly error! 
I, in general, have an alias `python` for `python3` in my `~/.bashrc` file. 
I commented this line out and ta-da: No errors

But what I do not understand is why does it work with other files, why did I never see this error. 
So, never mind!

daniel-thompson, 2020-05-19:
I think we're probably done with this issue for now. I believe all the questions about RLE 2-bit format have been explained above, #18 has been fixed and we've adopted a new palette for RLE-2bit images which is derived from the web-safe set and should be fairly easy to prepare greaphics for.

closed 2020-05-19 by daniel-thompson