wasp-os/wasp-os #16: Documentation - Flashing bootloader.hex

cr0mbly, 2020-05-02:
Hi Daniel awesome work on this I've just ordered my own PineTime and am excited to give this a spin.

I was planning on following [this](https://medium.com/@ly.lee/openocd-on-raspberry-pi-better-with-swd-on-spi-7dea9caeb590) guide to flash the `bootloader.hex` per your README.md does that make make sense to you in the context of Wasp-os are you able to update your docs with this or a better article explaining how this can be done if possible? It'd be handy for people just getting into this.

daniel-thompson, 2020-05-04:
I'm afraid this is something I decided not to work on (at least not yet)!

Two reasons, firstly I am fully occupied writing wasp-os and secondly documenting how to load binaries onto the PineTime isn't wasp-os specific and should be part of the PineTime documentation not part of the wasp-os documentation.

In fact that documentation is being starting to from on the wiki but, right now, it still needs a lot of work:
https://wiki.pine64.org/index.php/Reprogramming_the_PineTime . I could certainly add a link to this from the wasp-os documentation (perhaps accompanied by some text saying "it's a wiki... yes it is a bit rubbish right now... please make it better as you learn stuff" ).

PS I used two STLink devices to program my own watch... OpenOCD/stock-firmware to load softdevice and Black Magic Probe to load and debug the bootloader. This is a very weird way to do it which it the main reason I decided documenting how I did it wasn't going to be helpful! Of the currently documented approach using the RPi seems the best approach to me (and OpenOCD should automatically recognise a .hex file if you load that in place of the .elf files that the rust+mynewt scripts suggest).

daniel-thompson, 2020-05-06:
> Of the currently documented approach using the RPi seems the best approach to me (and OpenOCD should automatically recognise a .hex file if you load that in place of the .elf files that the rust+mynewt scripts suggest).

I guess you should also take a look at this thread on the forum:
https://forum.pine64.org/showthread.php?tid=8631

I guess I should add that rumours about "it broke when I did X" often swirl on forums and even the ones that are untrue stay around as a kind of superstition, sometimes for years afterwards (it's kind of like FUD but without any malice). In other words *please* don't take the above as "Daniel says bcm2835spi is dangerous", there's not really enough information available at this point to be sure. However I have to concede that, unless I already had two spare PineTime's on my desk, I personally would use the (older, slower but better tested) bcm2835gpio driver for the time being.

PS Many of the hook up guides for bcm2835gpio also include a 330 ohm resistor inline on the SWDIO line to prevent current flow if there are protocol errors.

commit 2f4f7282 (2020-05-20): README: Add a link to the Pinetime SWD Programming Guide

Fixes: #16
closed 2020-05-20 by daniel-thompson