wasp-os/wasp-os #19: (u)zlib missing

cool2man, 2020-05-06:
The first prototype of a png-8 viewer runs in the emulator, but couldn't run on a real device.

While the simulator has a zlib available (but no uzlib, which I would prefere because of the existing of the DecompIO function), on real clock neither the zlib, nor the uzlib is avalable. For me it looks like that the uzlib for micropython is compiled, but I have no idea, why it is not available.

Any help how this yould be fixed?

Thanks and regards

daniel-thompson, 2020-05-06:
Micropython is highly configurable and different ports can enable/disable different config options to achieve the best  balance of code size versus features.

By default the nRF port has most of the features turns off and this includes uzlib. I've enabled a few modules for wasp-os but if you want to use the zlib stuff you will have to enable it, see:
https://github.com/daniel-thompson/micropython/blob/5424de70d7cfb1d53828b54cd1df509c3f0fee74/ports/nrf/mpconfigport.h#L126

cool2man, 2020-05-07:
Thanks - this worked.

closed 2020-05-07 by cool2man