wasp-os/wasp-os #20: PGNViewerApp

cool2man, 2020-05-08:
First of all, I had no experience with Python before starting with testing wasp-os - so I am open for any input how my coding could be better ...

As a proof-of-concept, I wanted to write an app which displays a png-8 picture on the watch. I was not able to get it running on the real watch :-(

This is the app:

```
# SPDX-License-Identifier: LGPL-3.0-or-later
# Copyright (C) 2020 Daniel Thompson, Ralf Tossenberger
"""PNG-8 image viewer prototype
"""

import wasp
import icons
import sys
try:
    import uzlib
    used_uzlib = True
except:
    import zlib
    used_uzlib = False
try:
    import ubinascii
except:
    import binascii as ubinascii
import micropython

class PNGViewerApp():

    NAME = 'PNGVr'
    ICON = icons.app

    def __init__(self):
        """Initialize the application."""
        pass

    def foreground(self):
        """Activate the application."""
        self._draw()

    def background(self):
        """De-activate the application."""
        pass

    def sleep(self):
        """Notify the application the device is about to sleep."""
        return False

    def wake(self):
        """Notify the application the device is waking up."""
        pass

    def press(self, button, state):
        """Notify the application of a button-press event."""
        pass

    def swipe(self, event):
        pass

    def tick(self, ticks):
        """Notify the application that its periodic tick is due."""
        pass

    def touch(self, event):
        pass

    def _draw(self):
        """Draw the display from scratch."""
        hex = open("testbild.hex", "r")
        bin = open("testbild.png", "wb")
        for ln in hex:
            data = ln
            bin.write(ubinascii.a2b_base64(data))
        hex.close()
        bin.close()
        
        self.png8blit("testbild.png", 0, 0)

    def _update(self):
        """Update the dynamic parts of the application display"""
        pass
        

    def png8blit(self, file, x, y):
        """Draw a PNG-8 image."""
        
        with open(file, "rb") as png_file:
        
            assert png_file.read(8) == b"\x89\x50\x4e\x47\x0d\x0a\x1a\x0a" # PNG signature
            assert png_file.read(8) == b"\x00\x00\x00\x0d\x49\x48\x44\x52" # IHDR chunk with length
            png_width = int.from_bytes(png_file.read(4), 'big')
            png_height = int.from_bytes(png_file.read(4), 'big')
            assert png_width <= 240 and png_height <=240 # max dimensions
            assert png_file.read(5) <= b"\x08\x03\x00\x00\x00" # depth, type, compression, filter, interlace
            png_file.read(4) # skip crc
            
            chunk_len = int.from_bytes(png_file.read(4), 'big')
            while png_file.read(4) != b"\x50\x4c\x54\x45": # PLTE chunk
                png_file.read(chunk_len + 4) # skip chunk + crc
                chunk_len = int.from_bytes(png_file.read(4), 'big')
                assert chunk_len > 0 # PLTE chunk not found
            png_color_palette_size = int(chunk_len / 3)
            png_color_palette = [None] * png_color_palette_size
            for n in range(png_color_palette_size):
                png_color_palette[n] = int.from_bytes(b"\x00" + png_file.read(3), 'big')
            png_file.read(4) # skip crc
            
            chunk_len = int.from_bytes(png_file.read(4), 'big')
            while png_file.read(4) != b"\x49\x44\x41\x54": # IDAT chunk
                png_file.read(chunk_len + 4) # skip chunk + crc
                chunk_len = int.from_bytes(png_file.read(4), 'big')
                assert chunk_len > 0 # IDAT chunk not found
            png_file.read(2) # skip header
            if used_uzlib:
                png_file = uzlib.DecompIO(png_file, -15) # data without header and crc
            else:
                png_data = zlib.decompress(png_file.read(chunk_len - 2 - 4), -15) # data without header and crc
            
            display = wasp.watch.display
            display.set_window(x, y, png_width, png_height)
            buf = memoryview(display.linebuffer)[0:2*png_width]
            index_png_data = 0;
            
            display.quick_start()
            for py in range (png_height):
                if used_uzlib:
                    png_file_read(1) # skip first byte per line
                else:
                    index_png_data += 1 # skip first byte per line
                for px in range(png_width):
                    if used_uzlib:
                        rgb888 = png_color_palette[int(png_file.read(1))]
                    else:
                        rgb888 = png_color_palette[png_data[index_png_data]]
                        index_png_data += 1
                    rgb565 = (rgb888 & 0xff0000) >> 8 | (rgb888 & 0x00ff00) >> 5 | (rgb888 & 0x0000ff) >> 3
                    buf[px << 1] = (rgb565 & 0xff00) >> 8
                    buf[(px << 1) | 1] = rgb565 & 0x00ff
                display.quick_write(buf)
            display.quick_end()
```

Currently it’s not possible to upload binary data files, as a workaround I uploaded this image as a "hex" file via `wasptool --upload`

```
iVBORw0KGgoAAAANSUhEUgAAAPAAAADwCAMAAAAJixmgAAAAGFBMVEW1EBOztROzEbYRtLYPD7YQ
EBDs7O0TtRF+IEdIAAAA+0lEQVR42uzPsQGAIBAEMASF/TemgPY7u0s2SJthmrCwsLCwsLCwsLCw
sLCwsLCwsLCwsLCw8H/hJ4ywsLCwsLCwsLCwsLCwsLCwsLCwsLCwsLCwsLDwDY8wwsLCwsLCwsLC
wsLCwsLCwsLCwsLCwsLCwsLCVXiFERYWFhYWFhYWFhYWFhYWFhYWFhYWFhYWFhYWrsI9jLCwsLCw
sLCwsLCwsLCwsLCwsLCwsLCwsLCwcBUGAAAAAAAACPWGERYWFhYWFhYWFhYWFhYWFhYWFhYWFhYW
FhYWrsJfGGFhYWFhYWFhYWFhYWFhYWFhYWFhYWFhYWFhYeFjCzAAnh0IbqrdkrsAAAAASUVORK5C
YII=
```
and want to convert it inside _draw() (back) to a png image file.

png8blit should the read/unpack this png-8 file and send it to the display of the watch.

Inside the simulator all works fine.

I wanted to upload the app via `tools/wasptool --exec wasp/apps/pngviewer.py`, but the upload stops at somewhat over 80%. The watch does not respond anymore and has to be reset.

If I freeze the app, adding an entry in wasp/boards/pinetime/manifest.py and in wasp/wasp.py to register, the app could be started on the watch.

If I start the app, it crashed with an memory error inside png8blit(). It crashes with the same error, if I put inside this function, as the first statement, an `assert True = False`. For me it looks like, that his function needs to much heap (even without doing anything).

I am open for any ideas and have time to wait ...

daniel-thompson, 2020-05-11:
It's a shame this is running short of RAM, not least because I can't see anything obvious that would burn it up.

A couple of things worth noting:

1. I think there is a good chance that the asserts are being ripped out by the module freezing process so the assert True == False test may not actually be having any effect.
2. There is a typo (png_file_read instead of png_file.read) in the second line of the decode loop... so we can assume it never gets that far through the execution ;-) (otherwise you'd see a symbol error rather than a memory error).

Rather than asserting it might be better to call an undefined function (e.g. i_got_this_far() ) since that will give you a distinctive crash and can never be stripped out at the compile stage. This may allow you to explore how far you are able to execute the function. I'd suggest starting with a call just before (and then after) the call to uzlib.DecompIO() since maybe that is trying to allocate significant space for the dictionary.

If it does turn out to be constructing DecompIO() (or reading the first byte from it) that is causing the trouble then perhaps take a look at what compression level you are using for the PNG file... IIRC for lossless compression the compression level is often just a way to manipulate the dictionary size. That means that if your image exporter offers a compression level (gimp does this) then perhaps try setting that as low as possible whilst you are experimenting with the code.

cool2man, 2020-05-12:
You are right - asserts are complete ripped out. I used now `wasp.watch.vibrator.pulse() `to get in idea where the code is executed and where not.

After some tests before, I changed the first assert then to:

```
if png_file.read(8) == b"\x89\x50\x4e\x47\x0d\x0a\x1a\x0a":
    wasp.watch.vibrator.pulse()
```
I did not get a vibration, this means that the first 8 bytes are not as expected. Later on, there is a png_file.read(chunk_len + 4) which results then into an memory error `memory allocation failed, allocating 1229472855 bytes`. I expect that the chunk_len is much too high, because of the garbage data read before.

In case I had forgotten to upload the "hex file", I got the same! memory error. In that case, the PNG file should exists, but with a length of 0 bytes. So it seems, that the PNG file is not written.

I checked then the hex-to-bin-code. The `wasp.watch.vibrator.pulse()` inside the `for ln in hex:` gave me "the feeling" that the input file ist processed, but I have no idea what this routine has written (or not).

This is my current state, I will continue my checks in one of the next days.

daniel-thompson, 2020-05-12:
Keep in mind that one of the killer features of having a Python compiler/interpreter built into the watch is that you can fire up the console and test code fragments (like the hex decode) interactively.

There is also `from shell import *` which gives you a few commands to examine the filesystem: ls, cd('mydir'), cat('thing.hex') .

You could probably even hack together a quick bit of checksum code...

cool2man, 2020-05-12:
I forgot to use this features ...

```
>>> file_testbild = open("testbild.png", "rb")
>>> file_testbild.read(1)
    ERROR:dbus.connection:Exception in handler for D-Bus signal:
    Traceback (most recent call last):
    File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
    File "/home/coolman/wasp-os/tools/pynus/tealblue.py", line 260, in <lambda>
    self._signal_receiver = char_props.connect_to_signal('PropertiesChanged', lambda itf, ch, inv: self._on_prop_changed(itf, ch, inv))
    File "/home/coolman/wasp-os/tools/pynus/tealblue.py", line 273, in _on_prop_changed
    self.on_notify(self, changed_props['Value'])
    File "tools/pynus/pynus.py", line 42, in on_notify
    data = bytes(value).replace(b'\n', b'\r\n').decode('utf-8')
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0x89 in position 0: invalid start byte
```
It seems that the "b" of the file mode "rb" is ignored and the file is interpreted as a (utf-8) text file. `can't decode byte 0x89 in position 0` - yes, this is always the first byte of a png file.

daniel-thompson, 2020-05-12:
I hope you are, at least, _enjoying_ testing the limits (in ways I had only dreamed possible ;-) ).

This particular stack trace looks to me like a bug in the host tool that runs the console (and is also written in python). It has been sent a byte sequence by the target that has broken the utf-8 decoder.

Changing line 42 of tools/pynus/pynus.py should avoid the problem (by telling the decode() method not to throw an exception):

    data = bytes(value).replace(b'\n', b'\r\n').decode('utf-8', 'backslashreplace')

If you are interested you can see the difference in the following fragments of python (run on my workstation) and I figure the last behaviour is most useful to understand the messages from the target):

    >>> b'\x89'.decode('utf-8')
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0x89 in position 0: invalid start byte
    >>> b'\x89'.decode('utf-8', 'replace')
    '�'
    >>> b'\x89'.decode('utf-8', 'backslashreplace')
    '\\x89'

cool2man, 2020-05-12:
Ok, I changed pynus.py and can now check the code step by step.

```
coolman@coolgpd2:~/wasp-os$ ./tools/wasptool --console
Connected to PineTime (DA:61:81:6F:C8:05).
Exit console using Ctrl-X.

>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8)
'\x89PNG\r\n\x1a\n'
>>> png_file.read(8)
'\x00\x00\x00\rIHDR'
>>> int.from_bytes(png_file.read(4), 'big')
240
>>> int.from_bytes(png_file.read(4), 'big')
240
>>> png_file.read(5)
'\x08\x03\x00\x00\x00'
>>> png_file.read(4)
'\t\x8b\x19\xa0'
>>> int.from_bytes(png_file.read(4), 'big')
24
>>> png_file.read(4)
'PLTE'
>>> png_color_palette_size = int(24 / 3)
>>> png_color_palette = [None] * png_color_palette_size
>>> for n in range(png_color_palette_size):
...     png_color_palette[n] = int.from_bytes(b"\x00" + png_file.read(3), 'big')
...
>>> png_file.read(4)
'~ GH'
>>> int.from_bytes(png_file.read(4), 'big')
251
>>> png_file.read(4)
'IDAT'
>>> png_file.read(2)
'x\xda'
>>> import uzlib
>>> png_file_deflated = uzlib.DecompIO(png_file, -15)
Traceback (most recent call last):
  File "<stdin>", in <module>
MemoryError: memory allocation failed, allocating 32768 bytes
>>>
```
As you expected, uzlib.DecompIO needs too much memory!

Maybe to compression of the PNG file could be changed, but for me it makes more sense to stop investigating into PNG processing on the watch. At next I will try a mix of PNG reading (on PC) using the palette already inside a PNG with indexed color and mix this with rle-encoding.

I leave it open, why I got another memory message with the code direct running on the watch. But I leave not open that this code did not work as expected:

```
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8) == b"\x89\x50\x4e\x47\x0d\x0a\x1a\x0a"
False
>>> png_file.close()
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8)
'\x89PNG\r\n\x1a\n'
```
The byte array on the 2nd line ist the same as on the last - but I get a False on comparing?

daniel-thompson, 2020-05-12:
Interesting (really... I wrote the RLE encoder code partly because I didn't know enough about PNG format to even try to use it ;-) ).

W.r.t. the getting False on comparing it looks like you are comparing string with a bytestring. Does python3 on the host behave the same way?

cool2man, 2020-05-13:
running on watch:

```
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8)
'\x89PNG\r\n\x1a\n'
>>> png_file.close()
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8) == b"\x89\x50\x4e\x47\x0d\x0a\x1a\x0a"
False
>>> png_file.close()
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8) == "\x89PNG\r\n\x1a\n"
True
```

running in simulator

```
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8)
b'\x89PNG\r\n\x1a\n'
>>> png_file.close()
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8) == b"\x89\x50\x4e\x47\x0d\x0a\x1a\x0a"
True
>>> png_file.close()
>>> png_file = open("testbild.png", "rb")
>>> png_file.read(8) == "\x89PNG\r\n\x1a\n"
False
```

Who is right?

daniel-thompson, 2020-05-13:
I haven't built anything to test it but I think binary file support is another feature that is turned off by default. This is the code that handles the binary flag for the filesystem used on PineTime:
https://github.com/micropython/micropython/blob/18fb5b443200548c142d5aa83c08cf873f9524d3/extmod/vfs_lfsx_file.c#L71-L75

There I think the behaviour of the watch will change if you enable MICROPY_PY_IO_FILEIO.

cool2man, 2020-05-18:
I changed `png_file.read(8) == "\x89PNG\r\n\x1a\n"` to `png_file.read(8) == b"\x89PNG\r\n\x1a\n"` and the True / False answers worked as expected.

closed 2020-05-18 by cool2man