wasp-os/wasp-os #21: Good work!

kennethlimcp, 2020-05-13:
Just wanted to shout out that this is super cool. Makes me wanna buy a PineTime to play now! :D

cefn, 2020-05-16:
Indeed this is amazingly brilliant. I've been hoping for a Micropython on smartwatch project for a while!

I'm fairly experienced with Micropython on ESP8266 ( I run https://vgkits.org ) but I'm curious what I would need other than buying the Pinetime Smartwatch dev kit  to embark on simple Wasp-OS hacking and follow along. 

Could I potentially use one of these A6 devices for example... https://bit.ly/3bCpM6l

Do I need a physical device called an SWD programmer, or is it a software facility that uses a bluetooth adapter? If it needs hardware, what's a typical example of a device I could plug into a linux desktop to be able to flash Wasp-OS onto a PineTime or A6. 

Basically I couldn't work out if OTA bluetooth flashing (without a wired programmer) is an option or if this needs a hardware adapter in the beginning to flash it with a bootloader and hence enable OTA flashing.

Happy to work through whatever steps I can find but would be nice to know I have all the right stuff in the workshop before I attempt it and I can't quite work out the basic starter kit from the README.md

Thanks for your time if you're able to bullet a few links for a starter kit to buy.

daniel-thompson, 2020-05-16:
Once you get the wasp-bootloader installed on the watch then it can be updated OTA via Bluetooh (as shown in the videos) but the devkit won't come pre-programmed with a bootloader. There are lots of ways to install the bootloader (partially documented on https://wiki.pine64.org/index.php/Reprogramming_the_PineTime ) and if you need help with that sort of thing then I'd recommend asking on the PineTime forum: http://forum.pine64.org/forumdisplay.php?fid=134

If you're

jlukanc1, 2020-05-16:
I put together a repo with a small modified script to make flashing easier. However, it requires a Raspberry Pi. I have not been able to test it yet, but there should not be any danger.

[https://github.com/jlukanc1/pinetime-boot-flasher](Link to the repo)

closed 2020-05-17 by kennethlimcp

daniel-thompson, 2020-05-17:
Any chance that you can keep the wiki up to date with the best flashing methods? I'm going to start pointing the wasp-os docs to the wiki soon.

jlukanc1, 2020-05-17:
I don't have an ST-Link or any similar devices, other than the Pi. So I can't give any help for those. Although, the Pi may be a preferred option because of the ability to unlock flash and the ability to compile and such, it is also much more common.

cefn, 2020-05-17:
Thanks for getting back to me @daniel-thompson . Your comment at https://github.com/daniel-thompson/wasp-os/issues/21#issuecomment-629706370 seems to have been truncated, though. 

It finishes "If you're" 

I'm guessing from the comment that it's only really feasible to run Wasp-OS with a pinetime, then, and the ultra cheap watches like A6 are a no-go. 

From @jlukanc1 comment I could use an ST-Link but only after unlocking with the more complex/expensive setup which can execute an ERASEALL. 

Gather if necessary I could execute an ERASEALL via Raspberry Pi so thanks for your help documenting this.

SteveAmor, 2020-05-17:
@cefn you can run wasp-os on a P8 smart watch with a few pin changes. You can also OTA the wasp-os bootloader, without having to take the back off the P8 watch, using DaFlasher.

cefn, 2020-05-17:
Thanks, @SteveAmor that sounds much more approachable to get me started!

daniel-thompson, 2020-05-18:
> Thanks for getting back to me @daniel-thompson . Your comment at [#21 (comment)](https://github.com/daniel-thompson/wasp-os/issues/21#issuecomment-629706370) seems to have been truncated, though.
> 
> It finishes "If you're"

I think that was just half drafted text that had scrolled of the bottom of the comment box.

> I'm guessing from the comment that it's only really feasible to run Wasp-OS with a pinetime, then, and the ultra cheap watches like A6 are a no-go.

Not quite.

wasp-os can be ported to anything with an nRF52XXX in it (although it will be less work if you have a 240x240 display).

In other words, underneath the covers, I doubt there is much significant technical difference between PineTime and the A6. That, however, is missing the point! The difference between PineTime and other low end smart watches never was a technical: the difference is that the community around PineTime is much, much larger. There is a strong support forum, multiple different OS developers sharing ideas, reasonably good hardware documentation[1] and you don't need to remove the back of a test device with a hacksaw in order to get started[2] (because the case isn't glued shut when you buy it).

Or put more succinctly, if you want to enjoy the social aspects of open source development... get a PineTime ;-) there will be more people to talk to about it!

[1] Sure the wiki could be better... but it's a wiki... if it isn't good enough then you can make it better.
[2] DaFlasher is awesome BTW and it does means that if you buy *two* watches then you can probably spare one of them from the hacksaw. However most people will end up soft-bricking a watch they are developing on (especially if they want to add custom bootloader features) and to unbrick it will need to get the back off.

daniel-thompson, 2020-05-18:
BTW anyone looking at using the RPi SPI accelerated mode might like to take a look at this comment:
https://github.com/daniel-thompson/wasp-os/issues/16#issuecomment-624666270

cefn, 2020-05-30:
> @cefn you can run wasp-os on a P8 smart watch with a few pin changes. You can also OTA the wasp-os bootloader, without having to take the back off the P8 watch, using DaFlasher.

Is there a reference for the pin changes. I bought a P8a but haven't started flashing it yet. Expecting to brick it and would be useful to know how to unbrick.

SteveAmor, 2020-05-30:
@cefn here's the pinout https://cdn.discordapp.com/attachments/622348681538043924/688438175571771485/P8pinout.jpg