wasp-os/wasp-os #22: Black Screen after OTA-Update

tht-jxny, 2020-05-14:
Can you imagine why an application that is fully functional in the simulator results in a black screen on the actual watch when I uploaded the `micropython.zip`?

I now think that after some hints by you and some trial and error on my side I'll be able to fix any errors ^^. Thank you very much for your help so far!

daniel-thompson, 2020-05-14:
Can you go into a little more detail? For example is the whole watch gone black or just your application. Likewise if it is just your application then do the usually swipes to switch between applications still work. The more info the easier for me to guess what might be happening.

tht-jxny, 2020-05-14:
Sorry, I apologize. 
The whole watch stays black, there are no error messages and not even the start application that worked every time so far is now working.
Furthermore, I am not able to see any error messages on the screen.
I guessed that an import error might be the cause because in one application I have imported the `parser` module ( btw, which official modules can be used; Is the random library available) but the error still was there after I commented the app out.

Is there any option besides the one that I need to clone your repo again and after every change I have to build, upload and test the firmware?

SteveAmor, 2020-05-15:
@tht-jxny have you seen #14

tht-jxny, 2020-05-15:
> @tht-jxny have you seen #14 

I did. But I am not able to connect my watch to my Laptop via bluetooth ( Pairing refused). Therefore I only upload the firmware with my phone and cannot use the command line wasptool. 
If you have any ideas concerning my bluetooth problems, please share them with me.

SteveAmor, 2020-05-15:
@tht-jxny pairing is not required.  I find with wasptool that it can take a few attempts for it to connect to the watch.  Sometimes I reboot my Pi to try and connect again.  For OTA updates from my Pi, dfu.py is not as fussy as wasptool and usually connects first time.

daniel-thompson, 2020-05-15:
Just so we can be 100% clear rebooting and testing an application on a working device would give us the following:

1. Long press on the hardware button (~6 seconds) causes the device to reboot and show the bootloader logo (PineCone centred, "<- Start" in top-left, BT logo in bottom-right).
2. Short press on the hardware button (~half a second... but only *after* the bootloader logo has already been displayed for ~5 seconds) causes us the leave the bootloader and show the splash screen (PineCone centred, nothing else) for two seconds and then the screen is cleared.
3. Digital clock application appears.
4. Swipe up shows the launcher
5. Launching the application runs the appropriate draw code for the application.

If I have understood correctly the problems occur at 3. above. Instead of seeing the digital clock you see a black screen?

If so, then there can be a couple of reasons for that to happen. When the screen is black are you able to connect to the "PineTime" device using nRF connect (or an NUS client such as Serial Bluetooth Terminal)?

tht-jxny, 2020-05-15:
@SteveAmor I tried your advice:
Trying to connect with wasptool results in:
``` Shellscript
$ ./tools/wasptool  --console --device e8:4a:6e:af:d1:6d

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/wasptool", line 206, in <module>
    console.expect('Connect')
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f0b87492e48>
command: ./tools/pynus/pynus.py
args: [b'./tools/pynus/pynus.py']
buffer (last 100 chars): ''
before (last 100 chars): "elf._signal_receiver.remove()\r\nAttributeError: 'Device' object has no attribute '_signal_receiver'\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 13882
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Connect")

```

Then I tried uploading my firmware with dfu.py:
``` Shellscript
$ python3 tools/ota-dfu/dfu.py -f firmware.hex -d firmware.dat -a e8:4a:6e:af:d1:6d

    ================================
    ==                            ==
    ==         DFU Server         ==
    ==                            ==
    ================================
    
Sending file firmware.hex to E8:4A:6E:AF:D1:6D
Exception at line 141: name 'IntelHex' is not defined
DFU Server done

```

I once already tried `sudo hcitool lescan` and got the error `Set scan parameters failed: Input/output error`. I looked for solutions on Stackoverflow and so one but nothing worked for me so I then went with `bluetoothctl`.

@daniel-thompson Yes I still see the Pine64 logo but then the display turns black. I indeed can connect to the watch with nRF Connect - to my surprise. 

Any ideas?
Thanks for your help so far!

daniel-thompson, 2020-05-15:
For the wasptool problems try adding `--verbose` (and open a new issue with that info).

For the dfu.py stuff this looks like a missing dependency on the intelhex module for python3.

> @daniel-thompson Yes I still see the Pine64 logo but then the display turns black. I indeed can connect to the watch when the display is black with nRF Connect - to my surprise.
> 
> Any ideas?

That's good. Most likely that is just a python exception during startup. You need to use a special debug mode reset to find out what exactly is happening.

To enable the debug mode (which simple deliberately does not run the initialization code) you must be holding the hardware button down when the splash screen disappears (and for about a quarter second afterwards). Note that if you hold it down for 5 seconds the device resets so basically from the bootloader screen, wait 5 seconds then then press and hold the button and then release it a little bit after the splash screen disappears (this press will be about 2.5 seconds long ;-) ).

Once you have done that you need to connect to the python console (Serial Bluetooth Terminal is good for that if you can't get your laptop to work) and run the following commands:

    import wasp
    wasp.system.run(no_except=False)

Hopefully either this will work (in which case we need to change your main.py as @SteveAmor suggested) or it will give us a python stacktrace to help us debug.

SteveAmor, 2020-05-15:
@tht-jxny as you're not using wasptool for development, as a workaround, I assume you can add main.py to the manifest as suggested here: https://github.com/daniel-thompson/wasp-os/issues/6#issuecomment-598627679 and upload the new micropython build.

tht-jxny, 2020-05-16:
It worked with Serial Bluetooth Terminal. This is what I got:

``` Shellscript

Traceback (most recent call last):
18:33:07.249   File "<stdin>", in <module>
18:33:07.337   File "wasp.py", in <module>
18:33:07.389   File "apps/calc.py", in <module>
18:33:07.473   File "apps/calc.py", in CalculatorApp
18:33:07.521 AttributeError: 'module' object has no attribute 'calc'
18:33:07.608 >>> 18:33:10.004 wasp.system.run(no_except=False)
18:33:10.040 wasp.system.run(no_except=False)
18:33:10.403 Traceback (most recent call last):
18:33:10.444   File "<stdin>", in <module>
18:33:10.489 NameError: name 'wasp' isn't defined
18:33:10.537 >>> 
18:33:43.941 Connection lost
18:38:26.452 Connecting to PineTime ...
18:38:28.459 Connected
18:38:32.165 import wasp
18:38:32.283 import wasp
18:38:32.418 >>> 18:38:33.750 wasp.system.run(no_except=False)
18:38:33.767 wasp.system.run(no_except=False)
18:38:34.128 Traceback (most recent call last):
18:38:34.174   File "<stdin>", in <module>
18:38:34.218 AttributeError: 'module' object has no attribute 'system'
18:38:34.307 >>> 18:38:38.898 wasp.system.run(no_except=False)
18:38:38.943 wasp.system.run(no_except=False)
18:38:39.302 Traceback (most recent call last):
18:38:39.348   File "<stdin>", in <module>
18:38:39.395 AttributeError: 'module' object has no attribute 'system'
18:38:39.483 >>> 
18:39:06.857 Connection lost

```

As I never lack questions ans problems, here is the next one:
Are you sure that wasptool automatically choses the right colors of images when encoding them? I ran over the following:

```
...
File "draw565.py", line 55, in _fill
p[x] = color
IndexError: index out of bounds on dimension 1
```

daniel-thompson, 2020-05-16:
This is the bit that matters:

```
18:33:07.249   File "<stdin>", in <module>
18:33:07.337   File "wasp.py", in <module>
18:33:07.389   File "apps/calc.py", in <module>
18:33:07.473   File "apps/calc.py", in CalculatorApp
18:33:07.521 AttributeError: 'module' object has no attribute 'calc'
```

This is the backtrace where things go wrong (if we get an exception on the import then we can expect the next line not to work). It looks to me like there is no icons.calc . Try commening out the ICON = icons.calc line.

tht-jxny, 2020-05-16:
There indeed is a calc icon in the `icons.py` but I commented it out as you said. The same error message before, but this time I only get the latter part:
```
wasp.system.run(no_except=False)
18:38:34.128 Traceback (most recent call last):
18:38:34.174   File "<stdin>", in <module>
18:38:34.218 AttributeError: 'module' object has no attribute 'system`
```

What I found was [this](https://stackoverflow.com/questions/1250103/attributeerror-module-object-has-no-attribute) but for me it doesn't make sense in this case.

daniel-thompson, 2020-05-17:
> Are you sure that wasptool automatically choses the right colors of images when encoding them? I ran over the following:
> 
> ```
> ...
> File "draw565.py", line 55, in _fill
> p[x] = color
> IndexError: index out of bounds on dimension 1
> ```

Completely sure!

That looks more like #18 (which is now fixed... follow the links for the patch)

tht-jxny, 2020-05-17:
@daniel-thompson Okay. My image that I wanted to blit was nearly black and white and after I changed it to completely black and white everything worked. But thanks for mentioning the commit.
My intuition says that something is not right with my code - that there is the wrong content in the essential functions. Could you please look over my code either here or in my [repository](https://github.com/tht-jxny/MicroPineTime). And if you have enough time, could you please run my code on your device to make sure that the error is not related to my device.

daniel-thompson, 2020-05-19:
Works for me!

[image]

I copied wasp/apps/calc.py directly into my tree and merged in the extra icons into wasp/icons.py and rebuilt. I then sent some commands via the console (shown above) to register the app (which is basically just a hand coded version of the recommendations in the [application developer's guide](https://wasp-os.readthedocs.io/en/latest/appguide.html#freezing-your-application-into-the-wasp-os-binary).

Some general feedback:

- You might consider integrating the graphics into the application instead of icons (like wasp/apps/demo.py and wasp/apps/gameoflife.py do). icons.py is mostly there to make it easy to handle "generic" applications on devices without a 240x240 color screen but applications like the calculator will probably never be generic (they certainly won't run on a fitness band ;-) ).
- In general don't make untested code run automatically at power on... it is much easier to debug if you use the approach in the screenshot above (e.g. interrupt a "good" device and then import the untested code, you'll get much better error reports like that).

tht-jxny, 2020-05-26:
To finish things up:
Concerning my horrible workflow, I code on my Ubuntu 18.04 Laptop, flash the firmware with the nRF-Connect App and if there are any problems, I can get into the console with wasptool with my RaspberryPi.

I think the problem was that I imported a module which doesn't exist in micropython. I now fixed all errors and my code is available in my repo. 
Thanks a lot for all your recommendations and pieces of advice.

Since I'm familiar with developing mobile apps in Flutter, I now want to program a speedometer and connect it to my watch. Is there any documentation on bluetooth interaction? Receiving or sending bits of data? Will there be some in the future? Thanks for any replies!

closed 2020-05-26 by tht-jxny

daniel-thompson, 2020-05-27:
At the moment the bluetooth code is pretty basic (and it will change in future releases) so there's no documentation for it in the app writers guide.

All you have is the python console. This is not super robust but it should be enough to prototyping and experimentation. Basically get the speedo code on your phone to send something like this to the UART service on the watch. Assuming myapp is in the global namespace and gives the instance of your speedo app then the data to send is basically:

    ^Cmyapp.update_speed(33);wasp.system.run()

Note that the phone will also to read the reply characters from the above command also otherwise the watch may hang.

Needless to say there are plans to change how BT is managed but they won't be happening until I've got the step counter (and maybe also heart rate monitor) working.