wasp-os/wasp-os #23: Wasptool console: Input/Output error

tht-jxny, 2020-05-15:
I wanted to start the console with `./tools/wasptool  --console --device f0:35:a8:5b:04:b0 --verbose` with the error:
``` Shellscript
Traceback (most recent call last):
  File "MicroPineTime/tools/pynus/tealblue.py", line 392, in callback_wrapper
    callback(*args)
  File "./tools/pynus/pynus.py", line 50, in nus
    device = lookup_device(adapter)
  File "./tools/pynus/pynus.py", line 19, in lookup_device
    for device in adapter.devices():
  File "MicroPineTime/tools/pynus/tealblue.py", line 81, in devices
    yield Device(self._teal, path, properties)
  File "MicroPineTime/tools/pynus/tealblue.py", line 147, in __init__
    if properties['ServicesResolved']:
KeyError: 'ServicesResolved'
Exception ignored in: <bound method Device.__del__ of <tealblue.Device address=1C:A0:D3:99:00:29 name='SoundBuds Flow'>>
Traceback (most recent call last):
  File "MicroPineTime/tools/pynus/tealblue.py", line 156, in __del__
    self._signal_receiver.remove()
AttributeError: 'Device' object has no attribute '_signal_receiver'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/wasptool", line 206, in <module>
    console.expect('Connect')
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f7b8e9b1e48>
command: ./tools/pynus/pynus.py
args: [b'./tools/pynus/pynus.py']
buffer (last 100 chars): ''
before (last 100 chars): "elf._signal_receiver.remove()\r\nAttributeError: 'Device' object has no attribute '_signal_receiver'\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 20793
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Connect")

```
What am I doing wrong?

SteveAmor, 2020-05-15:
I've not used the device argument but it says it is a named device rather than the mac address. Try running the command without the device argument and it will connect to the first ble device it finds.

tht-jxny, 2020-05-16:
@SteveAmor Unfortunately, without the device argument it doesn't work either:
``` Shellscript
Traceback (most recent call last):
  File "MicroPineTime/tools/pynus/tealblue.py", line 392, in callback_wrapper
    callback(*args)
  File "./tools/pynus/pynus.py", line 59, in nus
    device = lookup_device(adapter)
  File "./tools/pynus/pynus.py", line 24, in lookup_device
    for device in adapter.devices():
  File "MicroPineTime/tools/pynus/tealblue.py", line 81, in devices
    yield Device(self._teal, path, properties)
  File "MicroPineTime/tools/pynus/tealblue.py", line 147, in __init__
    if properties['ServicesResolved']:
KeyError: 'ServicesResolved'
Exception ignored in: <bound method Device.__del__ of <tealblue.Device address=1C:A0:D3:99:00:29 name='SoundBuds Flow'>>
Traceback (most recent call last):
  File "MicroPineTime/tools/pynus/tealblue.py", line 156, in __del__
    self._signal_receiver.remove()
AttributeError: 'Device' object has no attribute '_signal_receiver'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/wasptool", line 224, in <module>
    console.expect('Connect')
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7fb25f099208>
command: ./tools/pynus/pynus.py
args: [b'./tools/pynus/pynus.py']
buffer (last 100 chars): ''
before (last 100 chars): "elf._signal_receiver.remove()\r\nAttributeError: 'Device' object has no attribute '_signal_receiver'\r\n"
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 24177
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Connect")
```
I assume that it is a hcitool error and I will try my best if I have time.

daniel-thompson, 2020-05-19:
> Exception ignored in: <bound method Device.__del__ of <tealblue.Device address=1C:A0:D3:99:00:29 name='SoundBuds Flow'>>

So... I finally spotted this (it has scrolled of the side of the github text box). pynus has selected the wrong device to attach to.

If you do a submodule update you will get a new version of pynus that introduces support for connecting to devices by MAC address (previous versions just did a scan and picked the first one that looked plausible). Unfortunately this is pretty new and wasptool hasn't integrated it yet so you will have to run it by hand (try: `tools/pynus/pynus.py --device PineTime` or `tools/pynus/pynus.py --address f0:35:a8:5b:04:b0`).

tht-jxny, 2020-05-19:
Both methods brought the sam disapointing result:
```
Traceback (most recent call last): 
File "MicroPineTime/tools/pynus/tealblue.py", line 392, in callback_wrapper
    callback(*args)
  File "tools/pynus/pynus.py", line 59, in nus
    device = lookup_device(adapter)
  File "tools/pynus/pynus.py", line 24, in lookup_device
    for device in adapter.devices():
  File "MicroPineTime/tools/pynus/tealblue.py", line 81, in devices
    yield Device(self._teal, path, properties)
  File "MicroPineTime/tools/pynus/tealblue.py", line 147, in __init__
    if properties['ServicesResolved']:
KeyError: 'ServicesResolved'
Exception ignored in: <bound method Device.__del__ of <tealblue.Device address=1C:A0:D3:99:00:29 name='SoundBuds Flow'>>
Traceback (most recent call last):
  File "MicroPineTime/tools/pynus/tealblue.py", line 156, in __del__
    self._signal_receiver.remove()
AttributeError: 'Device' object has no attribute '_signal_receiver'

```

SteveAmor, 2020-05-19:
Can you turn off all bluetooth devices except the PC and Pinetime so it has no choice but to connect to the Pinetime. It looks like it's trying to connect to some ear buds.

tht-jxny, 2020-05-19:
These specific earbuds were last turned on several months ago so, yes, all devices are shut down except my two PineTimes and my pc. But I will soon try to connect with my RaspberryPi to the watch. Maybe this will solve all my problems...

daniel-thompson, 2020-05-19:
> Both methods brought the sam disapointing result:

No worries... I think we are edging closer.

A quick experiment if you wouldn't mind (I won't have time to try this myself tonight). Try changing File "MicroPineTime/tools/pynus/tealblue.py", line 147 from:

    if properties['ServicesResolved']:

to

    if 'ServicesResolved' in properties:

tht-jxny, 2020-05-19:
I now also think we're reaching further. I changed this line and got a new error! Hurrah! ^^ Let's see:
```
python3 tools/pynus/pynus.py --address a1:b2:c3:d4:e5:f6
Connecting to PineTime (a1:b2:c3:d4:e5:f6)...
Traceback (most recent call last):
  File "MicroPineTime/tools/pynus/tealblue.py", line 392, in callback_wrapper
    callback(*args)
  File "tools/pynus/pynus.py", line 69, in nus
    if not device.services_resolved:
  File "MicroPineTime/tools/pynus/tealblue.py", line 212, in services_resolved
    return bool(self._properties['ServicesResolved'])
KeyError: 'ServicesResolved'

```

I also tried scanning by name which then resulted in the message `Scanning` which rested on my screen for a felt eternity.

tht-jxny, 2020-05-21:
So now I think I'm getting very close. I now use my RaspberryPi for PineTime development and it finally connects perfectly with the watch. One problem less.
But when I want to upload some test code to the watch I get a new `pexpect` error:
``` Shellscript
tools/wasptool  --exec wasp/apps/calc.py --eval "wasp.system.register(CalculatorApp())"
Preparing to run wasp/apps/calc.py:
Traceback (most recent call last):.................] 7.4%
  File "tools/wasptool", line 236, in <module>
    handle_exec(console, args.exec)
  File "tools/wasptool", line 131, in handle_exec
    paste(c, f, verbose, chunk)
  File "tools/wasptool", line 80, in paste
    c.expect('=== ')
  File "/home/pi/.local/lib/python3.7/site-packages/pexpect/spawnbase.py", line 344, in expect
    timeout, searchwindowsize, async_)
  File "/home/pi/.local/lib/python3.7/site-packages/pexpect/spawnbase.py", line 372, in expect_list
    return exp.expect_loop(timeout)
  File "/home/pi/.local/lib/python3.7/site-packages/pexpect/expect.py", line 181, in expect_loop
    return self.timeout(e)
  File "/home/pi/.local/lib/python3.7/site-packages/pexpect/expect.py", line 144, in timeout
    raise exc
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0xb6176690>
command: tools/pynus/pynus.py
args: [b'tools/pynus/pynus.py']
buffer (last 100 chars): "    b'\\xf4J\\x04H\\x80\\xac\\x81\\x05"
before (last 100 chars): "    b'\\xf4J\\x04H\\x80\\xac\\x81\\x05"
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 2703
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('=== ')
```

Any ideas?

daniel-thompson, 2020-05-22:
That looks like you have run out of RAM to parse the python code. You can double check this using the `--verbose` option.

There are some interesting (and well commented) hacks in wasp/apps/demo.py to avoid this problem. However it is possible that the CalculatorApp is too large to be parsed on the device and will have to be included in the manifest (when I tested it I included it in the manifest but I registered it by hand so I could see any errors loading it).

daniel-thompson, 2020-05-25:
Running of memory on the device when pasting code is now better handled by wasptool:
https://github.com/daniel-thompson/wasp-os/commit/60c48b447c036567d565a45c0ea6259baa713485

tht-jxny, 2020-06-07:
So to finish things up, I think the problem has something to do with the bluetooth module of my laptop or with hcitool.

Since this problem has little to do with this repo, we could close this issue.

I noticed something else with the wasptool console: After I flashed a new bootloader to my watch, connecting with `tools/wasptool --console` is not successful anymore. Have you changed something with the bootloader that somehow blocks a bluetooth connection?
Otherwise, feel free to close this issue! Thanks for all your help so far.

daniel-thompson, 2020-06-08:
> I noticed something else with the wasptool console: After I flashed a new
> bootloader to my watch, connecting with `tools/wasptool --console` is not
> successful anymore. Have you changed something with the bootloader
> that somehow blocks a bluetooth connection?

There have been no bootloader changes that should do anything like that. The only recent changes have been implementing pseudo-NVRAM and a few ergonomic changes to the UI. Neither should affect BT.

I sometimes find that after updating the firmware the first attempt to run wasptool after a micropython.zip upgrade fails however that usually solved itself on a second or third reconnection attempt (I've wondered if it depends on the timing of the scans by the host).

closed 2020-06-08 by daniel-thompson