wasp-os/wasp-os #27: Bootloop with nRF52-DK

joern-h, 2020-05-26:
Hi,
i got this development board with a fitting screen to experiment with from my school.
https://www.nordicsemi.com/Software-and-Tools/Development-Kits/nRF52-DK/Getting-Started

The board should be compatible with the PineTime, but its boot looping with the bootloader.
I am able to connect via BLE with nRF Connect for Android, but it reboots just after 3 seconds.

Do you have any resources that could help me debug the problem and develop a fix for it?
I would really like to try this firmware.

Best regards 
Jörn

daniel-thompson, 2020-05-26:
That's the watchdog firing.

Are you building BOARD=pinetime? If so that won't work because the watchdog code is trying to read a button state that the nRF52-DK doesn't have.

You could try BOARD=nitrogen but to be honest you probably want to try BOARD=pca10040 and when things break copy files from the nitrogen until things build OK.

joern-h, 2020-05-26:
Yes, i was building pinetime.

After fiddeling around a bit with different board i finally got a bootloader that was not boot looping.
I was able to upload a micropython zip via nRF Connect.
The zip that i had created same bootloop again. 

I saved the working bootloader, tried some different things and now even if i flash the same bootloader again it starts looping again...

Could you elaborate a bit on how nitrogen could help me?

I build it but the bootloader from it does not have image output and does not even have bluetooth functionality.

daniel-thompson, 2020-05-27:
> Could you elaborate a bit on how nitrogen could help me?

Simply that is expects the button to be connected to a different pin (which is pulled up on PCA10040 and therefore won't cause a reboot).

Either you must make your pca10040 system exactly like a pinetime (e.g. wire up a physical button identically to the pinetime schematic) or you should to a proper port to pca10040 which basically means going to each of the existing boards directories, copying one of the existing boards and then fixing up the pin definitions to match your development board. The main directories you need to look at to port to a new board are:

* Bootloader: https://github.com/daniel-thompson/wasp-bootloader/tree/wasp-os/src/boards
* Micropython (there is already a pca10040 port but you may have to enable extra features to match pinetime): https://github.com/daniel-thompson/micropython/tree/wasp-os/ports/nrf/boards
* Wasp-os: https://github.com/daniel-thompson/wasp-os/tree/master/wasp/boards

closed 2020-06-11 by daniel-thompson