wasp-os/wasp-os #29: Don't show a black screen when booting into safe mode

daniel-thompson, 2020-06-01:
wasp-os implements a special "safe mode" to allow easy recovery if the user provides a severely broken main.py . However since the safe mode shows a black screen it is also a bit of a land mine for anyone who launches into safe mode accidentally.

Why not modify the startup scripts to show some a distinctive icon when we are running in safe-mode?

SteveAmor, 2020-06-01:
Are there any instructions on how to enter safe mode?

daniel-thompson, 2020-06-01:
> Are there any instructions on how to enter safe mode?

Not yet... but it there is a troubleshooting guide on the [TODO list](https://wasp-os.readthedocs.io/en/latest/TODO.html). It will eventually be added to that.

Safe-mode is triggered by holding down the physical button as the pine cone disappears (go into the bootloader, hold down the button, keep it held through the reset and the pine logo and release it when the screen goes black).

daniel-thompson, 2020-06-12:
This is now fixed... safe mode doesn't result in a blank screen.

There is also exception handling enabled through most of the python boot process so if we get exceptions at boot the display will turn on and tell us what we were trying to run.

closed 2020-06-12 by daniel-thompson