wasp-os/wasp-os #30: Touch input not working

pebri86, 2020-06-04:
Hi, I just compile from source and touch input not working even on simulator. the simulator show clock app but could swipe to show other apps.

thanks.

SteveAmor, 2020-06-04:
Hi @pebri86 are you able to do a "git bisect" to identify which commit caused this issue?

pebri86, 2020-06-04:
> Hi @pebri86 are you able to do a "git bisect" to identify which commit caused this issue?

Yes, i will try to do a git bisect, did you know which commit the known good working?

daniel-thompson, 2020-06-04:
The simulator doesn't (yet) implement real swipe support... but it is simulated using the arrow keys.

pebri86, 2020-06-04:
> The simulator doesn't (yet) implement real swipe support... but it is simulated using the arrow keys.

Great, thanks for the information. Now I know that the simulator works.
Actually i managed to port wasp-os to P8 Smartwatch which identical specs with pinetime with different pinout for some peripheral like button, etc. Everything working e.g display, button etc including bootloader except for touchscreen.

i used pinout here for micropython board definition
```csv
SPI_SCK,P2
SPI_MOSI,P3
SPI_MISO,P4
NOR_CS,P5
I2C_SDA,P6
I2C_SCL,P7
ACCEL_INT,P8
LCD_DETECT,P9
TP_RST,P13
UART_TX,P11
BUTTON,P17
BL_LO,P14
MOTOR,P16
UART_RX,P15
DISP_DC,P18
USB_PWR,P19
BL_MID,P22
BL_HI,P23
P24,P24
DISP_CS,P25
DISP_RST,P26
P27,P27
TP_INT,P28
P29,P29
HRS_TST,P30
BATTERY,P31,ADC0_IN7
```
and for the bootloader pinout
```c
#define LEDS_NUMBER        0

/*------------------------------------------------------------------*/
/* BUTTON
 *------------------------------------------------------------------*/
#define BUTTONS_NUMBER     1
#define BUTTON_1           17
#define BUTTON_PULL        NRF_GPIO_PIN_NOPULL
#define BUTTON_ACTIVE      1
//#define BUTTON_ENABLE    15

/*------------------------------------------------------------------*/
/* UART
 *------------------------------------------------------------------*/
#define RX_PIN_NUMBER      15 /* not connected */
#define TX_PIN_NUMBER      11
#define CTS_PIN_NUMBER     0
#define RTS_PIN_NUMBER     0
#define HWFC               false

#define ST7789_SPI_DISPLAY

// Used as model string in OTA mode
#define BLEDIS_MANUFACTURER   "COLMI"
#define BLEDIS_MODEL          "P8"
#define BLEDIS_DEVICE_NAME    "P8DFU"

```
I got the the pinout from [Here](https://github.com/atc1441/ATCwatch/blob/master/ATCwatch/pinout.h) and confirmed it works on the watch with his fw.

Is there anythings i need to check to solve the issue?

Thanks

pebri86, 2020-06-05:
I tried debugging the driver last night, and found out that event[0] always return 0. The x and y ( event 1 and 2 ) are updated correspond the touch location.

SteveAmor, 2020-06-05:
@pebri86 I ported to a P8 and wasp-os works fine but I just noticed that I left TP_RST = P10.  That might be worth a try.  I know there have been some issues with touch on the arduino version of the P8 code.  My P8 watch stock firmware was almost likely MOY-TFK5-1.7.7

pebri86, 2020-06-05:
@SteveAmor My watch stock firmware also MOY-TFK5-1.7.7, in my case i need to do some initialization step on init driver then the controller working correctly. Also i implement interrupt handler on the driver (TP_INT P28), now the touch controller working nicely. Maybe i'll post an PR of my working port if the author will support on the P8 watch.
Thanks,

daniel-thompson, 2020-06-05:
I'd be very happy to get PRs.

I've been a bit lax until now (and haven't written a contributors guide yet) but wasp-os seems to be gaining attention (which is awesome) so I'll put that on the list! I will probably adopt [DCO](https://developercertificate.org/) and the Signed-off-by: model used for projects like the Linux kernel for further contributions since it is very lightweight and designed by people who know more than me about that kind of thing ;-).

If you are porting to a new watch that I don't own (and I don't have a P8 at present) then a well lit, high res photo of the watch sitting on a sheet of white paper and running something interesting would be very welcome too. As device support grows then decent photography and install instructions are key to making the project attractive to join. I will probably enable the github wiki to host device photos and install instructions.

pebri86, 2020-06-09:
Hi @daniel-thompson, 
2 days ago i finished update for the touchscreen driver for pinetime or p8 (basically using the same controller). Implemented interrupt and sleep mode made great impact to the battery life. Before, the watch only could lasted discharge 1 and half day, now with sleep mode implemented the device only discharged around 10% after using it on a full day with average use.

The documentation said to use sleep mode you need to reset the chip controller then write 0x03 to register 0xA5 to the touch controller chip. It seems the touch controller drained the battery when it's not in sleep mode.

pebri86, 2020-06-09:
I'll close this issue since i solve the problem already. I'll send PR once your Contributor Guide written.
Thank You

closed 2020-06-09 by pebri86

daniel-thompson, 2020-06-09:
I don't think you need wait for contributors guide. Other than adding "Signed-off-by: Name <email>" at the bottom of the commit message (which indicates you are certifying one of a, b, c, or d from the [DCO](https://developercertificate.org/) ) then everything else will be a normal github PR and code review.