wasp-os/wasp-os #31: Black screen after installing wasp-os on Pinetime

hrmckay, 2020-06-06:
I downloaded the wasp-os v0.2 files from the web page.  I used Openocd and an STlink to flash the bootloader.hex to the Pinetime.  The screen of the Pinetime showed the Pine64 icon, a Bluetooth icon, the word Start and an arrow pointing at the button. I then used an Android phone running the "nRF Connect" app to upload the micropython.zip file to PineDFU.  When the upload completed the Pinetime screen showed black.  The time and date etc. did not appear, even after pressing the button.

After a lot of testing sending commands to the Pinetime from the Android phone using the "Serial Bluetooth Terminal" app I found that sending the commands "import wasp" and "wasp.system.run()" would bring up the watch face, showing the time, date and battery icon.  It looks like the "main.py" program was not being run.  There was a "main.py" file in the /flash filesystem.

I issued commands from the phone to reformat the /flash filesystem. I created a "main.py" file in the /flash filesystem containing the commands "import wasp" and "wasp.system.run()".  I then held down the button until the watch rebooted. After the screen with the word Start appeared I pressed the button.  The watch face started as it should.

Why it works now and didn't on the first try I do not know.

SteveAmor, 2020-06-06:
@daniel-thompson should this line be commented out? https://github.com/daniel-thompson/wasp-os/blob/d62e6a22c7d37715553c6ce25dfdc483e93820ac/wasp/boards/pinetime/watch.py.in#L87

daniel-thompson, 2020-06-07:
Hi Hugh

Thanks for joining us here.

I've modified the start up script to include several delays and to say what it is running as it does so. This shouldn't fix the problem but hopefully the screen won't be black and we'll get a better understanding of what is going on.

[micropython.zip](https://github.com/daniel-thompson/wasp-os/files/4741506/micropython.zip)

daniel-thompson, 2020-06-07:
Hi @SteveAmor 

> should this be commented out?
> https://github.com/daniel-thompson/wasp-os/blob/d62e6a22c7d37715553c6ce25dfdc483e93820ac/wasp/boards/pinetime/watch.py.in#L87

That line isn't actually commented out, it is a macro processor command, see:
https://github.com/daniel-thompson/wasp-os/blob/master/tools/preprocess.py

SteveAmor, 2020-06-07:
Ah, I did wonder.
There's something odd here though.  I helped someone on the forums the other day with a similar issue.

bajansen, 2020-06-07:
I am having the same problem described in the original post. I uploaded the debug-enabled micropython.zip, and instead of displaying a black screen, the watch now shows 'Run main.py'.

SteveAmor, 2020-06-07:
Are you able to look at the contents of main.py on your flash? If you can console in, try
```
import shell
shell.cat("main.py")
```

bajansen, 2020-06-07:
`>>> shell.cat("main.py")
# SPDX-License-Identifier: LGPL-3.0-or-later
# Copyright (C) 2020 Daniel Thompson

import wasp
from apps.patch import ColorPatchApp
wasp.system.register(ColorPatchApp())
wasp.system.run()`

Based on this information, the problem appears to be in the apps.patch import. When I run this through the console, this module cannot be found.

`>>> from apps.patch import ColorPatchApp
Traceback (most recent call last):
  File "<stdin>", in <module>
ImportError: no module named 'apps.patch'`

daniel-thompson, 2020-06-08:
@bajansen : Thanks so much for following up with this.

It looks like there is a mistake in the generated binaries (but not in the source). The code you have found main.py was prototype code when I added the slider widget to wasp-os. It looks this was incorrectly included in the wasp-os-0.2 binaries.

I will regenerate them this evening but that will not fix the problem for you. main.py is intended to be user configurable (it is what allows you to make your own applications run at boot time if  you want them to) so once the default version has been written to flash wasp-os will continue to run it.

To fix the problem on your side you need to have the following contents in main.py:
~~~
# SPDX-License-Identifier: LGPL-3.0-or-later
# Copyright (C) 2020 Daniel Thompson

import wasp
wasp.system.run()
~~~

If you have wasptool working then you can replace main.py using `./tools/wasptool --upload  main.py`. If you are using a different console tool then try `import shell` and `shell.upload('main.py')` and follow your nose.

bajansen, 2020-06-08:
I figured as much, seeing the main.py file in the source tree did not include the problematic import. I uploaded a main.py file without the problematic code and have just finished OTA of the non-debug micropython.zip. The watch now correctly displays the watchface after booting.

hrmckay, 2020-06-08:
I tried the debug micropython.zip and it works great.  Six messages come out one after the other and then the watch face appears (since I fixed my main.py).

daniel-thompson, 2020-06-11:
This is fixed as of https://github.com/daniel-thompson/wasp-os/releases/download/v0.2/wasp-os-0.2-remastered.tar.gz .

closed 2020-06-11 by daniel-thompson