wasp-os/wasp-os #32: Errors building from source

hrmckay, 2020-06-09:
I followed your instructions on building from source and encountered some errors.  I am running Debian Buster under Parallels on a Macbook Pro.  I forgot to add the Arm toolchain to the PATH but I fixed that.  I got an error on the "make -j 'nproc' BOARD=pinetime all" command until I realized 'nproc' should be the number of processors.  The other errors I could not fix.

I got a syntax error in hex2c.py while building bootloader.

I got the error "Failed to import ecdsa, cannot do signing" while building micropython.zip which I hope is a warning message.

I got an invalid syntax error while trying to build sim.

Attached is the console log.

[pinetime-build.txt](https://github.com/daniel-thompson/wasp-os/files/4749684/pinetime-build.txt)

daniel-thompson, 2020-06-09:
> I followed your instructions on building from source and encountered some errors. I am running
> Debian Buster under Parallels on a Macbook Pro. I forgot to add the Arm toolchain to the PATH but
> I [...]

There is a spelling error here. The correct invocation is: ``make -j `nproc` BOARD=pinetime all`` (backticks not single quotes).

> I got a syntax error in hex2c.py while building bootloader.

hex2c.py uses [f-strings](https://docs.python.org/3/whatsnew/3.6.html#whatsnew36-pep498 ) which is a python language feature first introduced in Python 3.6 (and will hopefully be implemented in MicroPython very soon). To run hex2c.py you will needs python >= 3.6 .

Python 3.6 has been out for a long time and Debian Buster should include Python 3.7. Forgive my asking but are you sure you are running Debian Buster (your prompt says parallels@debian9 which would be Debian Stretch).

> I got the error "Failed to import ecdsa, cannot do signing" while building micropython.zip which
> I hope is a warning message.

Correct. There is no need to worry about this.

> I got an invalid syntax error while trying to build sim.

This is another f-string problem. If you can fix the build I think this will probably go away too.

closed 2020-06-09 by daniel-thompson

daniel-thompson, 2020-06-09:
I updated the docs based on this thread and the Fixed: tag made this issue auto-close... please do feel free to reopen if you have any more questions.

commit dea2ba8d (2020-06-09): README: Document python version requirements

Fixes: #32