wasp-os/wasp-os #35: Bootloop after upgrading from afb9bd16b6c811f1a211e3a159b57aa1a787073f

purlupar, 2020-06-11:
I get a bootloop (Pine logo - short black - again Pine logo - etc.) after flashing the most recent wasp-os. I guess it's because of the bootloader not being updated. How can one flash the bootloader (using dfu-tool or nrfTools, which both failed for me)? Thanks!

daniel-thompson, 2020-06-12:
Ignore this post... just looking for better ways to reference a gitsha.

- ... from afb9bd16b6c811f1a211e3a159b57aa1a787073f
- ... from afb9bd1 ("wasp: Add support for brightness caching and keep-awake signalling")

daniel-thompson, 2020-06-12:
This is very unlikely to be bootloader related, the [changes since Jan](https://github.com/daniel-thompson/wasp-bootloader/commits/wasp-os) are all backward compatible (one wilkl. Problems after flashing are normally caused by a bad flash and are solved with a reflash.

However if you want to update to the latest bootloader updating to the latest bootloader is done by loading reloader.zip instead of micropython.zip. When the reloader is launched you will see a blue pinecone that is changed to white to indicate progress and then the system will reset back to the bootloader (ready to reinstall micropython.zip).

If the back of your watch is glued on be very careful with reloader.zip. Early versions have absolutely no error checking at all... so a bad flash could brick the device. The latest version is much more robust with pre-flash checksum checks and post-flash consistency checks... but the code to do the checks is *very* new (IIRC I checked it in yesterday).

Does it boot loop if you try to run in safe mode (when leaving the bootloader keep holding down the button until slightly after the pinecone disappears then release).

purlupar, 2020-06-12:
Thanks for your reply (and formatting the post) -- I'm afraid I still get the same bootloop after having flashed the lastest reloader.zip and then the latest micropython.zip. I haven't achieved more with holding the button longer as you indicate, for safe mode. What would do you next if this was your watch? :-) Thanks.

daniel-thompson, 2020-06-12:
I would try things like:

- Test micropython.zip from the latest available binary release (see https://github.com/daniel-thompson/wasp-os/releases/tag/v0.2 )
- Check I was using a new enough compiler (although it you have built older versions of wasp-os successfully then you probably already are)
- Check that going back to an older build works correctly (and it is does use git bisect to hunt down the problem).
- Removing boot.py from wasp/boards/pinetime/manifest.py and rebuilding (this should give you a black screen but will disable almost all the init code... but the console should still be available via the UART service to help you figure out what is going on).

daniel-thompson, 2020-06-12:
Oh... almost forgot... and if the back isn't glued on (and maybe even if it was) I'd try debugging via SWD.

purlupar, 2020-06-12:
Thanks for these suggestions -- the first one (flash a released binary) worked, it boots! Closing for now, I'll post here if I can find out what's going on through the console.

closed 2020-06-12 by purlupar

purlupar, 2020-06-12:
I tried to build everything from the point v2 was released, that is I checked out f56cf0e.
Then this happens:

make -j `nproc` BOARD=pinetime all
...
python3 -m nordicsemi dfu genpkg \
	--bootloader bootloader/_build-pinetime_nrf52832/pinetime_nrf52832_bootloader-*-nosd.hex \
	--softdevice bootloader/lib/softdevice/s132_nrf52_6.1.1/s132_nrf52_6.1.1_softdevice.hex \
	**bootloader.zip**
Usage: __main__.py dfu genpkg [OPTIONS] ZIPFILE
Try "__main__.py dfu genpkg --help" for help.

**Error: Got unexpected extra argument (bootloader.zip)**
GEN build-pinetime-s132/frozen_content.c
make: *** [Makefile:19: bootloader] Fehler 2
make: *** Es wird auf noch nicht beendete Prozesse gewartet....
CC build-pinetime-s132/frozen_content.c
LINK build-pinetime-s132/firmware.elf
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: /tmp/firmware.elf.N0DFGy.ltrans2.ltrans.o:(.rodata+0x209c): undefined reference to `mp_module_bma42x'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:429: build-pinetime-s132/firmware.elf] Fehler 1 # (error 1 in German)
make[1]: Verzeichnis „/home/david/wasp-os/micropython/ports/nrf“ wird verlassen
make: *** [Makefile:40: micropython] Fehler 2

Checking out master, this does not happen. I don't understand what's going on here. Do you perhaps?

purlupar, 2020-06-13:
Hello, 

> I would try things like:
>     * Test micropython.zip from the latest available binary release (see https://github.com/daniel-thompson/wasp-os/releases/tag/v0.2 )

As mentioned, this does work. So does flashing a micropython.zip which I had built in March.
>     * Check I was using a new enough compiler (although it you have built older versions of wasp-os successfully then you probably already are)
I remember it works exactly with the arm-none-eabi-gcc version you mention (9-2019-q4 major). I downloaded it again (in case I had accidentally updated it to a newer version via my OS's package manager). How do I tell the build process to use that one though? (IIRC during the installation of micropython, is that correct?)

> 
>     * Check that going back to an older build works correctly (and it is does use git bisect to hunt down the problem).
See above. Seems the problem is not even wasp but build-tools related, perhaps

>     * Removing boot.py from wasp/boards/pinetime/manifest.py and rebuilding (this should give you a black screen but will disable almost all the init code... but the console should still be available via the UART service to help you figure out what is going on).
Did not get so far, yet (see above).

purlupar, 2020-06-14:
Ok, turned out I was using not the gcc-arm mentioned, but a newer (I think) one. I can build now, awesome!

https://github.com/daniel-thompson/wasp-os/issues/35#issuecomment-643288910