wasp-os/wasp-os #37: Firmware too big for device

tht-jxny, 2020-06-17:
I have finally done it. I broke the system. ^^ Yesterday, I wanted to build my firmware and got this:
``` Shellscript
Zip created at build-pinetime/reloader.zip
make[1]: Leaving directory 'wasp-os/reloader'
mv reloader/build-pinetime/reloader.zip .
gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: build-pinetime-s132/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 13500 bytes
collect2: error: ld returned 1 exit status
Makefile:428: recipe for target 'build-pinetime-s132/firmware.elf' failed
make[1]: *** [build-pinetime-s132/firmware.elf] Error 1
make[1]: Leaving directory wasp-os/micropython/ports/nrf'
Makefile:35: recipe for target 'micropython' failed
make: *** [micropython] Error 2
```

I now want to optimize my code concerning memory consumption but I think what matters the most storage wise is the bma-driver. Any other ideas?

daniel-thompson, 2020-06-17:
The main ways to reduce the footprint are to switch to a smaller softdevice (micropython has currently adopted S132 but I think it would still work OK with S112) and to move resources (graphics, fonts, BMA config file) to the filesystem on the SPI flash. It would also be worth following up on the TODO in micropython/port/nrf/boards/memory.ld that mentions the filesystem size.

daniel-thompson, 2020-06-17:
a02bd3b follows up the TODO I mentioned in my last comment. It should free up and extra 64K of FLASH.

commit a02bd3bc (2020-06-17): micropython: Disable micro:bit filesystem

... and enable persistant code loading.

Fixes: #37
closed 2020-06-17 by daniel-thompson