wasp-os/wasp-os #39: Problems with icons

hrmckay, 2020-06-22:
I want to add a new icon to a watch face I am building, a simple Bluetooth icon.  I used a png file of a 32x32 white Bluetooth icon on a transparent background.  I used your tool rle_encode to encode it and  rleblit to display it.  On the simulator the display did not look like it should, see the attached screenshot (icon is at left on same line as June 21).  I tried using blit, same problem.  I tried using rle_encode with --2bit but there was an error, see attached console log.  I tried a different icon, a png with a 30x30 white Bluetooth icon on a black background.  Same problems, except the --2bit encode did not fail.  The icons are attached.  Any idea what the problem is?

[rle_encode_error.txt](https://github.com/daniel-thompson/wasp-os/files/4815381/rle_encode_error.txt)
[image]
[image]
[image]

daniel-thompson, 2020-06-23:
I think the initial problems are linked to alpha. The two RLE formats currently used by wasp-os do not support alpha channels.

The second problem is that the supplied image cannot be encoded as a 1-bit RLE image since that format supports only black and white (and the .png has lots of gray pixels in it). The image looks like it is working using the 2-bit encoder. Does it display correctly?

hrmckay, 2020-06-23:
Daniel, thanks for the quick reply.  The Bluetooth icon with the black background does display correctly with 2-bit encoding.  I enlarged that icon in a picture editor and it does have a lot of gray pixels for a simple white icon on a black background.  I will make a note to myself about no transparent backgrounds,

Thanks,

Hugh

closed 2020-06-23 by hrmckay