wasp-os/wasp-os #40: Black screen on boot

Raveh00, 2020-06-23:
Hi,

I flashed the latest version of the bootloader and I was able to upload micropython.zip with nRF connect.
But after the bootloader I get a black screen.
I tried to debug it based on the other issues but I'm stuck.

If I run the following commands it works but not after a reboot:
```
import wasp
wasp.system.run()
```

So I tried to find the main.py but the filesystem doesn't seems to be mounted.

I found on the forum to look for the flash chip, it's the same as the one already reported:
XT25F32BWOIG
1924HTC
E96035

But there is an error when running this command:
```
>>> from watch import * ; flash = FLASH(spi, (Pin('NOR_CS', Pin.OUT, value=1),))
from watch import * ; flash = FLASH(spi, (Pin('NOR_CS', Pin.OUT, value=1),))
Traceback (most recent call last):
  File "<stdin>", in <module>
  File "flash/flash_spi.py", in __init__
  File "flash/flash_spi.py", in scan
ValueError: negative shift count
```

And effectively when I try to read the manufacturer ID I get a buffer full of 0:
```
>>> buf = bytearray(6) ; mvp = memoryview(buf) ; mvp[:] = b'\0\0\0\0\0\0'
buf = bytearray(6) ; mvp = memoryview(buf) ; mvp[:] = b'\0\0\0\0\0\0'
>>> cs = Pin('NOR_CS',Pin.OUT,value=1)
cs = Pin('NOR_CS',Pin.OUT,value=1)
>>> mvp[0]=0x9f ; print(buf) ; cs(0) ; spi.write_readinto(mvp[:4],mvp[:4]) ; cs(1) ; print(buf)
mvp[0]=0x9f ; print(buf);cs(0);spi.write_readinto(mvp[:4],mvp[:4]);cs(1);print(buf)
bytearray(b'\x9f\x00\x00\x00\x00\x00')
bytearray(b'\x00\x00\x00\x00\x00\x00')
```

Any idea what could be the issue?

daniel-thompson, 2020-06-24:
Firstly I'm a little curious what version of wasp-os you are using. A week or two ago I added some new code to try and show more informative messages if something goes wrong during boot. For example if there was a ValueError when constructing the FLASH object then the watch should display something like "Init SPINOR   FAILED" instead. Given the debug you've done I would not expect the latest version to help much but we shouldn't be getting black screens if the flash isn't working (and if you have a few minutes to try that I'd be very grateful... error messages are much more "search engine friendly" than black screens).

After updating to the latest version you could try:

1. Try the patch below. It adds some (very paranoid) sleeps to make sure the wake up command is correctly delivered. As mentioned these changes are paranoid so are unlikely to be a fix but worth checking "just in case". The other change is to try booting the OS anyway even if there is no filesystem (this code path is untested but might work ;-) ).

2. Let the battery run completely flat (the Torch app can be good for that). If the flash is in a "funny" state then this will hopefully be clears when the regulator shuts down. Note that I am aware some people simulate a flat battery by briefly short circuiting the battery to forcefully reset all the board components... let's just say that is at your own risk.

~~~
diff --git a/wasp/boards/pinetime/watch.py.in b/wasp/boards/pinetime/watch.py.in
index d98c99d8f569..43cc567fcdc9 100644
--- a/wasp/boards/pinetime/watch.py.in
+++ b/wasp/boards/pinetime/watch.py.in
@@ -89,9 +89,13 @@ try:
     # Release flash from deep power-down
     boot_msg("Wake SPINOR")
     nor_cs = Pin('NOR_CS', Pin.OUT, value=1)
+    time.sleep_ms(10)
     nor_cs(0)
+    time.sleep_ms(10)
     spi.write('\xAB')
+    time.sleep_ms(10)
     nor_cs(1)
+    time.sleep_ms(10)
 
     # Mount the filesystem
     boot_msg("Init SPINOR")
@@ -122,3 +126,7 @@ try:
 except:
     drawable.string("FAILED", 0, 136, width=240)
     backlight.set(1)
+    time.sleep(2)
+
+    import wasp
+    wasp.system.run()
~~~

Raveh00, 2020-06-24:
Hi Daniel,

My battery discharged completely last night and now everything works well.
So yeah the flash was certainly in a "funny" state.

But I don't know why I didn't had the error message.
I've downloaded this archive yesterday:
https://github.com/daniel-thompson/wasp-os/releases/download/v0.2/wasp-os-0.2-remastered.tar.gz
And uploaded micropython.zip extracted from it.

I didn't tried https://github.com/daniel-thompson/wasp-os/files/4741506/micropython.zip for issue #31 if it's the one you are talking about.

Thanks for your help.

daniel-thompson, 2020-06-24:
No worries. The recent changes to avoid "black screen" came after v0.2 was released.

closed 2020-06-24 by daniel-thompson