wasp-os/wasp-os #41: Stuck on bootloader (PineTime)

makedin, 2020-06-28:
This might be a bit of a general help request, best posted on the Pine64 forum, but I have been unable to register a user account there so please bear with me.

I have been able to flash the bootloader file on my PineTime (using Raspberry Pi and OpenOCD). As far as I can tell, this part is completely successful. The next step then is to upload micropython.zip using either nRF Connect on an Android device or the ota-dfu tool on a Linux computer. I have tried both, multiple times, but with limited success: every time the upload appears to succeed but as the device restarts I'm greeted by the bootloader screen.

Pressing the hardware button will cause the device to restart (sometimes, anyway; I haven't really been able to determine what kind of press does the trick) and brings back the bootloader screen. Sometimes the screen goes black for a little longer than usually, but the end result is always the same. When the device is in this state (i.e. always, unless I have erased the flash to get the screen to turn off) it will accept a new DFU connection, but wasptool (unsurprisingly) fails to find what it's looking for.

I have built the `bootloader.hex` and `micropython.zip` files from a couple of different commits, most recently 7b3e0f9 and also tried the prebuilt files from `wasp-os-0.2-remastered.tar.gz`. None of this makes any difference that I can discern. The sim seems to work just fine, but then I guess it doesn't actually use the firmware file anyway? I can't say for sure that my PineTime is completely healthy, but I did spend a little time poking playing with the software that came preinstalled and everything seemed to work.

Am I missing something or might there be a real problem needing further investigation? I have basically no experience using OpenOCD for this kind of thing so I'd be very grateful for any pointers.

hrmckay, 2020-06-28:
With the Pinetime watch, if you press and hold the button for five or six seconds it will reboot and show a screen with the Pine64 icon, a Bluetooth icon, the word Start and an arrow pointing at the button.  Wait several seconds. I usually wait at least 10, then press the button and release it.  Sometimes this button press doesn't work for me and I have to try again. The screen should go black, then the Pine64 icon reappear, then the watch screen with the time, date and battery icon should appear.

Hugh

makedin, 2020-06-29:
Thanks Hugh! It seems, though, that my watch behaves differently from yours: I get to the bootloader/OTA updater screen (logo, text, arrow, Bluetooth icon) as soon as the thing wakes up. Pressing the button again will always have the same effect: the screen goes black and a few seconds later the previously described screen comes back. This happens after a short button press, and a longer one appears to have no further effect.

daniel-thompson, 2020-06-29:
I'm afraid all I can really do at the moment is confirm your suspicions: yes, the simulator doesn't run the firmware file and no, you are not missing anything obvious.

The normal behaviour after flashing is to jump directly into the payload that we just flashed (IIRC without any splash screen). So the expected behaviour after burning micropython.zip is the pinecone disappears and then the clock appears after a short delay ("short" meaning not-much-more-than-a-second).

As mentioned in the README there can be some slightly odd effects from not cleanly disconnecting the SWD programmer but I think these are mostly related to power management rather than whether or not we can reboot correctly. Unfortunately I didn't use an RPi for initial programming so I can't really comment on how to cleanly disconnect! Hopefully ordering OpenOCD to exit *before* removing the hookup wire will do the right thing.

Button behaviour should be relatively consistent. A long press (~5 seconds) will reboot into bootloader mode, a short press when '<- Start' is displayed will try to launch the payload. This can do one of several things:

1. Working correctly: Pine cone logo for 2 seconds, black screen for 1 second, clock comes up
2. Firmware fails validation check: Pine cone logo for 2 seconds, <- Start and the BT logo appear
3. Firmware is corrupt (but passes simple checks): Pine cone logo for 2 seconds, black screen of 5 seconds, Pine cone logo for 2 seconds (again), <- Start and the BT logo appear
4. Firmware experiences an early boot problem: Pine cone logo for 2 seconds, black screen (or in recent versions a text string saying which part of early boot is faulty)

makedin, 2020-06-29:
Thanks for the input! It's good to hear that I do have a problem other than my own incompetence (though I wouldn't have minded an easy solution either).

The `exit` command is what I have been using for (ostensibly) disabling the debug interface. As I was writing this though, I finally found how it should really be done; the instructions are on the [wiki page for PineTime](https://wiki.pine64.org/index.php/PineTime) – not in the programming article. Trying to carry out those instructions seems to have either the watch or OpenOCD even less cooperative than perviously. Here's hoping I can get back to where I was earlier.

The errors I saw this time (now safely out of the bounds of my terminal emulator's rollback buffer) together with some of the other things I've seen trying to program different binaries into my PT make me suspect that the flash memory on my device might be misbehaving.

The behavior I'm seeing is consistent with the number 2 of your list. I wonder if it would be a large undertaking to add some debug output to the bootloader to get a better idea of where things go wrong? I took a quick look at the source code but couldn't immediately tell where to start (GitHub's famously terrible search functionality might have had something to do with this – guess I'll do some grepping on my local copy).

SteveAmor, 2020-06-29:
@makedin have you waited for the battery to go flat to see if it works properly after a short time charging?  That could resolve a state problem with the SPI flash.
I wouldn't worry about 'cleanly disconnecting'. I've never had a problem just removing the SWD cable.  The instructions on the wiki are to disable debug to reduce the current consumption which happens anyway when the battery goes flat.

makedin, 2020-06-30:
@SteveAmor I think the battery has gone flat during this process: at one point the screen started to go black very soon after removing the SWD-cable. I assume the cable supplies enough power to keep the watch going, but haven't really tried to verify it. I've since done a great many things to the device so whatever good it did has probably been undone since.

Is the idea to flash the bootloader and then let the battery go flat before sending `micropython.zip` over?

SteveAmor, 2020-06-30:
@makedin I've only ever SWD flashed the bootloader, then OTA DFU micropython.zip. One time it didn't run the python code until the battery went flat and was put on charge.  
If you have 3.3v on the SWD cable, it can power the watch.  Mine will even put some charge in the battery.
The fact that you have the bootloader working is a good start.  Did you see the fault finding tips in #40

daniel-thompson, 2020-06-30:
> The behavior I'm seeing is consistent with the number 2 of your list. I wonder if it would be a large undertaking to add some debug output to the bootloader to get a better idea of where things go [...]

Try searching for "led_state".

This is useful because many of the important transitions of bootloader state result in led_state() being called so its good for helping you navigate.

It is also useful because the bootloader uses led_state() to update the display (PineTime has no LEDs so I modified the LED infrastructure to use the update the screen instead). The code is pretty ugly but you should be able to change to colour of the pinecone to reflect whatever led states you are interested in (or decide to add).

Another thing for you to look at is:
https://github.com/daniel-thompson/wasp-bootloader/blob/wasp-os/src/main.c#L213

I think this is probably the test that is failing and maybe you can instrument to find out why.

Finally remember that you can use SWD to dump the contents of flash to a file. Sometimes it will be quicker to look at the flash state in a hex editor than to instrument when the only output you have is a multi coloured pine cone!

makedin, 2020-06-30:
Note: I wrote most of this post while @daniel-thompson was writing his, which is why I only respond to it at the very bottom. I decided to leave the rest of the text as it was.

I had seen #40 but had perhaps been a bit too hasty to dismiss it as a different problem (the reporter did after all get wasp-os running). I did let the battery go flat just now, but the original problem still persists.

I say original, since I observed an interesting phenomenon in the interim: after I had tried to disable the debugging mode (as mentioned previously) trying to perform the DFU upload caused the watch to enter a loop similar to the item number 3 in @daniel-thompson's post – the difference being that it never entered the "<- Start and the BT logo" -screen on its own. Pressing the button would invoke it, but only for a couple of seconds, after witch it was back to the "5 seconds dark, 2 seconds Pine cone logo" loop. The only way out of _that_ situation was to re-flash the bootloader. And this was not a one time thing either, it only stopped happening after the battery went flat.

I'm more and more convinced that this is not a wasp-os specific problem, but I think there are some interesting lessons for a bootloader developer to be learned here: 1) problems with the flash memory (if that's what we are looking at) can manifest in more ways than seem to be documented and 2) it is possible, at least in some circumstances, to get the watch in a (possibly previously unseen) broken state by merely performing a DFU upload of a known good `micropython.zip` file. Should keep life from getting too boring!

@daniel-thompson now that is what I was hoping for! Displaying a multi colored pine cone sounds like printf-debugging without all the annoying textual output. Guess I'll also figure out how to dump that flash, but I feel I already learned something interesting with just vandalizing the bootloader code (namely, the problem doesn't seem to be associated with bootloader validation. I'll have to poke around some more to be sure though).

makedin, 2020-07-01:
Having done some more digging I have now learned a few more things. I'll leave some notes in here in the hope that someone else can make any sense of what I've observed.

`bootloader_app_is_valid(DFU_BANK_0_REGION_START);` returns true. In fact, all the relevant checks in `main()` return what we'd like them to. There seems to be nothing blocking the app from starting. Double reset detection does what it's supposed to (or at least I think it does. It definitely does _something_).

In order to ignore as many moving parts as possible, I placed some code right after the app validity check, namely:
```c
if (valid_app) {
    board_teardown(); //also tried without this line
    if ( !sd_inited ) softdev_mbr_init();         
    bootloader_app_start(DFU_BANK_0_REGION_START);
    NVIC_SystemReset();
  }
```
Removing all the carefully placed checks in the code seemed like a good idea, but (totally surprisingly) it didn't work. The result is a boot loop. Well, at least I have some idea of where I should be looking next. I also get the feeling that it might soon be time to start reacquainting myself with ARM ISA.

I'd also like to note that pine cone debugging has proven to be absolutely delightful. Not terribly productive, to be sure, but very pleasing visually. Blinkenlights for the 21st century.

daniel-thompson, 2020-07-01:
Whilst it would be nice if you were having fun and the watch was working properly at least 50% is better than nothing!

I'll be disconnected from the internet from now until middle of next week but I'll look forward to seeing how things have gone when I get back.

PS It might be time to add instrumentation early in the micropython main method... probably best to use hardware breakpoints for that through... when it takes five minutes to flash over BLE then pinecone/backlight debugging won't be much fun.

SteveAmor, 2020-07-04:
@makedin is this the same as #42?  How old is you PineTime?

makedin, 2020-07-05:
@SteveAmor very possibly! My PT is fairly new: I received the shipping confirmation email just over a month ago, having placed my order three-ish weeks earlier. The delay between the two events made me wonder if it might not be from a new batch, manufactured sometime between mid May and early June, but I weren't able to find any reports of people having a problem like mine so I kind of dismissed the thought. There is a solder blob on the 5V pad.

Trying gdb debugging definitely seems like the next reasonable step, seeing how I've started encountering heisenbugs with pine cone debugging and I feel that the possibilities with that method have now been pretty much exhausted.

Unfortunately though, I've been unable to establish an SWD connection since I flashed a vanilla(ish) bootloader a few hours ago, making it impossible to test the hypothesis. The output from OpenOCD is what you'd expect when the SPI wires are not properly connected, but I'm _quite_ sure they are. Also, when the data wire is floating, OpenOCD reports a random SWD DPIDR, which is not happening now. It's not like the watch is dead however: it happily runs the bootloader and accepts DFU uploads – not that it does me much good.

I should also note, while I still remember, that at one point when the watch was being even less cooperative than usual, I decided to check the memory protection register/bit (`nrf52.dap apreg 1 0x04`) and it had flipped back to 0. That's not supposed to happen, is it? Setting it back to 1 got me back to the state I've previously described.

SteveAmor, 2020-07-06:
@makedin sometimes my RPi with OpenOCD stops responding but a reboot gets it working.  It is probably worth going back to a known state with your PT.  I would suggest:
1. Download precompiled V0.2 files
2. Flash Erase PT
3. SWD precompiled V0.2 bootloader.hex
4. OTA-DFU precompiled V0.2 micropython.zip

BTW what version is your gcc arm compiler? From #40, V10.1.0 appears to cause issues.

makedin, 2020-07-07:
@SteveAmor all, good suggestions, thanks!

Unfortunately my problem seems to run a bit too deep for that procedure to be applicable: I can't get past the second step of your instructions. I've rebooted my RPi multiple times during this process and let the PT battery go flat several times as well. Seems to make no difference. The behavior I'm seeing is distressingly similar to what's described in [this thread](https://forum.pine64.org/showthread.php?tid=8631). Guess I'll try once more with a new RPi in the off change that's where the problem lies, but then I'll be just about out of ideas.

The point about GCC version seems like an important one: I've been using v10.1.0. That could explain the trouble I've had with my debug builds, but it's not like things were working any better with the prebuilt files either.

daniel-thompson, 2020-07-10:
Hi @makedin 

Sorry to hear you are now doubly stuck. I do have a couple of new PineTime on the way so hope to be able to use these to reproduce some of the recently reports problems switching away from the stock firmware (or at least be able to offer a "works for me" based on experience of more than a single device).

Regarding the SWD just stopped working problem I'm interested in following that up since it is, at the same time, a commonly reported problem (e.g. enough to worry me) but also one where the problem reports lack details (e.g. we can't really follow up or draw a conclusion). In particular what driver are you using (bcm2835spi, bcm2835gpio, ...) and how were things hooked up (raw wires or safety resistors)?

PS *If* the problem comes about because of damage to a pad driver we don't even know whether the damage is at the RPi end or the PineTime end (arguably the RPi is the more fragile of the two devices). In other words an experiment you could try is switching to bcm2835gpio driver and choosing some different pins on the RPi connector.

makedin, 2020-07-10:
Hi, @daniel-thompson, good to hear from you again. Your suggestion regarding trying a different driver is an interesting one, and I'll try it just as soon as I can figure out how it's done. Before that though, I'll tell you what I can about the SPI problem.

I have been using the "real", non-bit bang SPI interface. I used the raw wires that came with the watch; I only read about the need to use current limiting resistors after I'd already gotten myself into this situation. I was able to reprogram, flash erase or just connect to the watch maybe a couple of dozen times before the problem suddenly manifested. [Here](https://pastebin.com/tWA0y2La) is the output of an OpenOCD run with `--debug` flag. If there is something else I should try in order to get some more useful intel on this, just let me know and I'll get to it.

SteveAmor, 2020-07-10:
I never got around to adding resistors and haven't had issues.  Are you sure your programming cable doesn't have a break?

hrmckay, 2020-07-10:
When I first got my PineTime I used a Raspberry Pi 4 to remove the flash protection.  I had a difficult time doing this, I kept getting error messages on the Pi indicating there was no connectivity to the watch.  The wires looked like they were connected to the watch but it kept failing.  I removed the wires, used a multimeter to check that they were not broken, put them back, still failed.  Then I thought maybe it is not the watch end but the Pi end.  I wiggled the connectors on the wires connected to the Pi and reran the unlock code and it worked.  One (or more) of the wires had an intermittent problem on the end that connects to the PI (or the STLink).  After that whenever anything failed I would wiggle those wires and retry.  I was happy when I could do updates over Bluetooth.

Hugh

daniel-thompson, 2020-10-13:
I think its time to close this one.

All the original "black screen" issues have been fixed and the SPI NOR wake up code has been rewritten (twice).

There is now a single way to black screen (corrupt main.py). We know what it means and how to help people who experience it *and* it cannot affect a first time user ;-) .

closed 2020-10-13 by daniel-thompson