wasp-os/wasp-os #42: Segfault

sqrtroot, 2020-07-04:
After flashing the bootloader and then dfu-ing the firmware i'm getting segfaults when the app starts.

I'm using a bluepill as a blackmagic probe. This gives me gdb access. Loading the symbol files with add-symbol-file (the bootloader.out for the bootloader and firmware.elf for micropython) gives me the following backtrace.
```gdb
(gdb) bt
#0  memset (s=0x20003a78 <impure_data+168>, c=0, n=1776) at ../../lib/libc/string0.c:85
#1  0x00026000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Do you have any idea what would cause this crash, is it something wrong with micropython, or the setup for the pinetime?

SteveAmor, 2020-07-04:
Could you provide details of the version (or commit) of the bootloader and micropython.zip and what happens on the watch display.  And just to confirm if it is a PineTime watch.

sqrtroot, 2020-07-04:
Yes it's a pinetime watch.
I'm running the master branch for wasp-os which has bootloader @ commit `9148e76` and micropython @ commit `d11200432`.

The watch boots into the pine icon, black screen rinse and repeat. If i hold the power button I can get into dfu-mode, and run dfu-updates.

SteveAmor, 2020-07-04:
@sqrtroot this is happening to quite a few people. How old is your PineTime?

sqrtroot, 2020-07-04:
I got my PineTime 1 day ago. There was also an issue with the charging pad, (the positive pad was resoldered on).
The hardware id of the nrf52832 in my pinetime is _0x0139_. Could you check the hwid of your device, in gdb you can do this by running 
```gdb
mon read hwid
```

The device info from `mon read deviceinfo` gives me this:
```
Part:    NRF52832
Variant:  AAE0
Package:  QF
Ram:    64K
Flash:    512K
```

SteveAmor, 2020-07-05:
I have a blob of solder in the 5v charging pad and others do too, so nothing to worry about.  
I am not able to check the hwid.  
Are you using the recommended gcc arm version - GNU-RM toolchain (9-2019-q4) from Arm?

sqrtroot, 2020-07-05:
I'm running `arm-none-eabi-gcc (Arch Repository) 10.1.0`.

SteveAmor, 2020-07-05:
@sqrtroot have you tried the V0.2 release rather than compile yourself?

sqrtroot, 2020-07-06:
I've compiled V0.2 and that also didn't work. I haven't checked if it's failing in the same function.

SteveAmor, 2020-07-06:
Can you try the pre compiled V0.2?  There's a link at the bottom of the repo homepage.

sqrtroot, 2020-07-06:
Okay, this version seems to be working, would my newer compiler be a problem somehow?
The other problem I have now is that it keeps disconnecting me, but that also might be my laptop.

SteveAmor, 2020-07-06:
Yes, I would say that it is most likely the version of compiler.

daniel-thompson, 2020-10-13:
I think its time to close this one.

Compiler versions can matter and there was also a bug with dependency tracking in this kind of time frame that has been fixed long since. If the pre-processor failed for any reason and you didn't notice the error message (amoung the nearby deluge) than the error became sticky and resulted in broken builds.

closed 2020-10-13 by daniel-thompson