wasp-os/wasp-os #43: SPINOR flash exception (negative shift count)

wbjeffrey, 2020-07-09:
I just received my pinetime this week. I'm trying to get wasp-os to work, but I ran into the following rather interesting problem. I should first state that I tried the precompiled 0.2 build, but only got a blank screen after the pinecone bootloader.

I pulled the master yesterday and built it. It stops with 'Init SPINOR FAILED'. 

After quite some hours of instrumenting code to narrow this down, I've gotten it down to a ValueError exception being raised with 'negative shift count' as the specific problem. This is happening during the call 'flash = FLASH(spi, (nor_cs,))'. The first shift I see is in the scan() method where the manufacturer id and flash size is being read.

I'm not terribly familiar with this HW yet, so I'm posting this to see what you think.

I've tried catching this exception, skipping the flash FS work, and forcing on safe_mode, but the code never seems to proceed further. I haven't gone down that road to see why yet. I'm stuck with the rather slow 5 minute micropython.zip flashing turnaround time.

SteveAmor, 2020-07-09:
When you flashed the precompiled V0.2 release did it stay on a black screen after the pinecones bootloader? If so, you should be able to connect to the REPL which might be useful to fault find using Python.

SteveAmor, 2020-07-09:
Also have a look at #6 which is similar but should have been fixed in the code now.

wbjeffrey, 2020-07-09:
@SteveAmor to question #1, yes, V0.2 stayed on a black screen. But I think by building and installing the later code we are getting to the real issue I'm having. 

To question #2, this looks exactly like my problem, except the code in the master does the wake-up call prior to trying to work with the flash. This leaves me with apparently non-working flash.

I did test the watch as fully as I could with the factory firmware before starting down this path and everything seemed to be working correctly. I don't suppose there is some easy way to power down this thing short of waiting for the battery to die is there? With the battery soldered in, I am not really excited about my options there.

I'll give your suggested, possible workaround a try and see how far it takes me.

daniel-thompson, 2020-07-10:
Hi @wbjeffrey 

Take a look at https://github.com/daniel-thompson/wasp-os/issues/40#issuecomment-648744637 . Basic on the debug you have done (negative shift usually happens if we read back all zeros from the ID register) I think you are seeing the same thing. The reporter for #40 found the problem "just went away" when the battery was flat and flash got reset. However in their case the battery ran out by accident, so I don't think they ever had the chance to try the experimental patch I shared (that patch is a long shot but I'd be interested in knowing the result since I can't reproduce this problem).

Regarding full power down... some people simply short out the battery for a short period of time but I've never tried that myself! Other than that I'm afraid there's only waiting (the torch program can be good for draining the battery) or desoldering.

Daniel.

wbjeffrey, 2020-07-10:
@SteveAmor  Once I added main.py back into the manifest file as suggested, I get an operational watch. I missed that part originally as a fix for skipping the flash filesystem.

@daniel-thompson So, I've tried every variation (including variable timings) on the release from deep power down that I can think of. None worked for me. I've also tried a software reset on the chip and a combination of the two. Nothing I've tried has worked. Either my flash chip is really broken, or it is in a state only a power cycle can fix. I would really have preferred to find a robust software based solution, but unless someone can suggest something else to try, I'm thinking of letting the battery run down to see if that fixes the problem.

SteveAmor, 2020-07-10:
@wbjeffrey you can turn the flashlight app on to flatten the battery :)

wbjeffrey, 2020-07-11:
So, after running the battery down, the flash is working. While I'm glad I have functional HW, this looks to be a concerning condition. I don't suppose the HW has any kind of power control we can enable eventually? I could see this situation becoming a recurring problem for people. I wonder if it might become bad enough as more watches get into more hands that it would be worth a HW revision to allow a mechanism for full power down/restart?

daniel-thompson, 2020-07-11:
I have a couple of additional PineTime on order and one of the reasons for me to order spares it so I can hook up an logic state analyzer to find out what commands used by the proprietary firmware to wake and sleep the flash. Hopefully that will give us enough clues to put this one to bed for good.

AlwaysLivid, 2020-08-02:
Had this issue myself-- I just used some barely functioning cables, rebooted a few times somehow. Then, it randomly started working. I used the latest bootloader/micropython.zip files.

wbjeffrey, 2020-08-19:
@daniel-thompson I've figured out a reliable way to reproduce this problem (at least on my watch), and I think I have a reliable fix. To reproduce this problem and get the 'Init SPINOR FAILED' message, use the wasptool console, ctrl-c, and execute the following:

import shell
import os

from machine import Pin
from machine import SPI
from flash.flash_spi import FLASH

nor_cs = Pin('NOR_CS', Pin.OUT, value=1)
spi = SPI(0)

# sleep flash
nor_cs(0)
spi.write('\xB9')
nor_cs(1)

After doing this, a long button press to reset to the boot loader, and then a press to boot, should consistently give you the init spinor failed problem. It does for me. 

Modifying watch.py as follows seems make the firmware wake the flash and work consistently:

    # Release flash from deep power-down
    boot_msg("Wake SPINOR")
    nor_cs = Pin('NOR_CS', Pin.OUT, value=1)
#   nor_cs(0)
#   spi.write('\xAB')
#   nor_cs(1)
    nor_bufp = bytearray(6)
    nor_mvp = memoryview(nor_bufp)
    nor_RDP = 0xAB
    nor_mvp[:] = b'\0\0\0\0\0\0'
    nor_cs(0)
    while nor_mvp[0] != 0x15 and nor_mvp[1] != 0x15 and nor_mvp[2] != 0x15:
        nor_mvp[:] = b'\0\0\0\0\0\0'
        nor_mvp[0] = nor_RDP
        nor_mvp[1] = 0x01
        nor_mvp[2] = 0x20
        nor_mvp[3] = 0x30
        spi.write_readinto(nor_mvp[:4], nor_mvp[:4])
    nor_cs(1)

This is adapted from a looking at issue #47 and a suggestion from MirkoCovizzi to look at the ATCWatch project.

Hopefully this works for you as well, and can lead to consistent sleep/wake on the flash and maybe improved battery life.

wbjeffrey, 2020-08-19:
Ugh. Should have previewed. The comments in the pasted code went to bold titles. Hopefully the edit fixes it.

daniel-thompson, 2020-08-19:
Interesting (and excellent). Do you know how many times it goes around the loop? I'd like there to be a limit otherwise we'd never get a FAILED message if the hardware actually is broken. Also I'd be interested to know if the loop works with == 0 rather than != 15 (to make the wake up code more portable).

This definitely looks likely to help us optimize battery life. Ideally both the sleep and wake up code can be integrated into wasp/drivers/flash/flash/flash_spi.py .

Do you want to develop the idea into a pull request or do you want someone else to finalize this? It is certainly OK to put the code into watch.py.in for now (which will fix the Init SPINOR FAILED bug) and we can look into generalizing this and integrating it into the flash driver a little later.

wbjeffrey, 2020-08-19:
I don't know how many times it loops. I must admit, the possibly infinite loop makes me uncomfortable too. The logic is an exact translation of the C/C++ code in the ATCWatch flash driver. I didn't want to tinker with it until I was sure it worked correctly. I'll instrument it and find a good limit threshold to avoid the infinite loop. I put this here to see if you thought it was worth more effort and how you would like to see it integrated.

Let me take a crack at building some functions to sleep and wake the flash in the flash_spi.py driver. I think that is the best long term place for it too. I'll try to get it put into a pull request in the next few days (as I get spare time). At least it is documented here as a working starting point.

wbjeffrey, 2020-08-20:
@daniel-thompson I've modified the flash driver and the watch.py.in and tested. You have 2 pull requests: 1 in wasp-os, and 1 in micropython-eeprom.

daniel-thompson, 2020-10-13:
Closing... the PRs have been absorbed (thank you!).

closed 2020-10-13 by daniel-thompson