wasp-os/wasp-os #44: wasp not included in micropython

mulm, 2020-07-11:
I built from scratch from commit c739df119b06b59b26c8f0c314d99fb453646f28.  Bootloader works, am able to load the built micropython.zip using wasptool, but wasp application does not seem to be included.  After bootloader bootsplash and micropython (pine logo) bootsplash, nothing appears on screen.  micropython seems to be working fine since I was able to reload micropython.zip using `./tools/wasptool --ota micropython.zip`.

However, somehow `wasp` is not included:
```
$ ./tools/wasptool --verbose --rtc
Connected to PineTime (E2:8C:3F:11:D1:C8).
Exit console using Ctrl-X.

>>> print("ERP""SHX")
print("ERP""SHX")
ERPSHX
>>> watch.rtc.set_localtime((2020, 7, 11, 14, 32, 8, 5, 193))
watch.rtc.set_localtime((2020, 7, 11, 14, 32, 8, 5, 193))
>>> wasp.system.run()
wasp.system.run()
Traceback (most recent call last):
  File "<stdin>", in <module>
NameError: name 'wasp' isn't defined
```

Did my build somehow fail?

mulm, 2020-07-11:
FYI, this is how I built (following guide):
```
git clone https://github.com/daniel-thompson/wasp-os
cd wasp-os
make submodules
make softdevice
make -j `nproc` BOARD=pinetime all
```

I also tried starting with a fresh clone with this `make` invocation, just in case some wasn't correct in Makefile:
```
make -j 1 BOARD=pinetime all
```

SteveAmor, 2020-07-11:
Hi @mulm have you tried the precompiled Release V0.2?  What version of gcc arm are you using?

SteveAmor, 2020-07-11:
It might start working if you let the battery drain and it does a hard reset.

mulm, 2020-07-11:
I pushed a bunch of .py files to the device and watch application started working.  Then, reloaded micropython.zip OTA and it still worked.  Not sure what actually fixed it.

How do I see the contents of the filesystem, as mentioned in #40?  Could that have been empty but now has applications?  Or could I have had a stale main.py or something in the filesystem due to previously having loaded wasp-os (many months ago)?

SteveAmor, 2020-07-11:
To see the list of files on the SPI flash you can use the micropython shell module.  Start the console from wasptool:
```
>>> import shell
>>> shell.ls()
>>> shell.cd("/flash")
>>> shell.ls()
>>> shell.cat("main.py")
```

mulm, 2020-07-11:
I removed all files found w/ `shell.ls()` and rebooted using long press on button.  Black screen on reboot.

Then, I tried:
```
./tools/wasptool --upload wasp/main.py
```
and rebooted using button and now device seems as-expected.  So, perhaps micropython.zip as built from master as of today does not include a correct main.py?  I would think the main application would work "out of the box".

SteveAmor, 2020-07-12:
The issue is that if you already have a main.py (from an older build) it doesn't get overwritten.
There is a test that checks you can mount the SPI flash. If you can't, it formats the flash and writes main.py to it. 
Perhaps there should also be a test to check if main.py exists and if it does, that the content is correct for the version of wasp-os.

daniel-thompson, 2020-07-12:
main.py is the tool for user customization of the device giving them control of the device before it has booted in order to add/remove applications. I'm not sure I want to get into heuristics to determine if the user's main.py is up to date or not.

Currently main.py did change recently but anything from (remastered) 0.2 and onwards should work OK (although you will get extra features if you update to latest main.py). However if upgrading from before 0.2 (the first binary release) then it is important to update main.py or things may not work. I'm planning to document the importance of updating main.py during upgrades (and how to do it) before the 0.3 release.

mulm, 2020-07-12:
Once I removed all files from /flash and reprogrammed micropython.zip, I would think this would be the same completely wiping the device.  And in that state, my device booted to black screen and required pushing the default main.py .  That seems to indicate to me that main.py is/was somehow missing from my micropython.zip.

Is a main.py _supposed_ to be included in the pinetime micropython.zip on a fresh clone or not?  I would think that would be part of a "great developer experience".

@daniel-thompson's last post seems to suggest that my problem was due to previously loaded code.  But, I'd like to make _sure_ that is the case...  is there an easy way to verify that the out of box experience for a first time user of wasp-os on the dev kit is to boot to the normal watch application?

I am clear about how to proceed with my own device and my own development, so I am otherwise ready to close this issue.

daniel-thompson, 2020-07-13:
The code is here: https://github.com/daniel-thompson/wasp-os/blob/6686f17e724b9a22318ff2248ce126a97dd19db4/wasp/boards/pinetime/watch.py.in#L104 . Currently it should behave exactly as Steve summarized:
> There is a test that checks you can mount the SPI flash. If you can't, it formats the
> flash and writes main.py to it. If you can mount the SPI flash, it assumes that
> everything is ok and it moves on.

mulm, 2020-07-13:
Thanks for holding my hand through this!  I appreciate the helpful answers so I actually learn something rather than just closing this issue with minimal explanation.

_However_, I do that might still be a real, but maybe minor issue...  doesn't all of the above mean that the problem might occur for any user who was transitioning either from pre-0.2 firmware (my problem it seems) _or any other code_ (like one of the many other pinetime firmware variants) _that has formatted the flash_ with a micropython compatible filesystem?

Any developer able to answer the question above can feel free to close this issue if either my logic is incorrect or this is not a serious issue.

daniel-thompson, 2020-11-08:
I think I'm going to close the issue at this point.

main.py is still a bit awkward for users who are not aware of it... but over the last few months there is much more documentation starting to give clues about how it works.

closed 2020-11-08 by daniel-thompson