wasp-os/wasp-os #46: Wasptool: Timeout - Not able to get command prompt

jeliebig, 2020-07-28:
Hello,

I just installed the firmware on my PineTime and tried to access the console using wasptool. But everytime I try to connect to my PineTime, my watch goes black and nothing happens. I can't even start it after that and need to force a reboot (5sec button press and 2sec button press to boot to micropython again).

I installed both the current master version and v0.2 on my PineTime but both gave me the same results.

Maybe this is has something to do with the bluetooth driver of my raspberry pi 4. I can't upload the firmware using dfu.py because it will always fail with "No Notification received". But uploading via the android app works without problem. So I think maybe there is something wrong with my bluetooth driver, but as I am not sure I want to post this issue to hear if you have some ideas about that.

Besides that issue, my watch works without any problems and the self test runs smoothly. Thank you for your work!

Looking forward to contributing as soon as I get the console to work.

Julian

hrmckay, 2020-07-29:
I am having similar problems with Bluetooth and a Raspberry Pi.  I have a Pi4 running the beta 64 bit Raspbian.  Uploading the firmware from the Pi sometimes works and sometimes fails, at different points in the upload but always with the same error message (see the dfutool_error.txt attachment).  The Wasptool program always fails,  whatever the command (see the wasptool_error2.txt attachment).

Hugh
[dfutool_error.txt](https://github.com/daniel-thompson/wasp-os/files/4991800/dfutool_error.txt)

jeliebig, 2020-07-29:
Yup, that's the first error message I got. So I adjusted parts of the dfu modules to show the traceback and fixed this error message. (I just added string.decode() to the if comparison) 

But after this message is fixed you will just get a connection error as well. 

Hope this little text can help somehow.

hrmckay, 2020-07-29:
I should have thought of running it with sudo, after all hcitool requires sudo.  I ran wasptool --rtc with sudo and it still failed but with a different error (Timeout).  Here is the log.

Hugh
[sudo_wasptool_error.txt](https://github.com/daniel-thompson/wasp-os/files/4997024/sudo_wasptool_error.txt)

SteveAmor, 2020-07-29:
You can add the ```--verbose``` argument to wasptool which might help get to the bottom of this.  Also note that it will try to connect to the first BLE mac address it sees unless you supply it as an argument.

jeliebig, 2020-07-29:
Sorry might have been too vague in my first message. I already ran it with the verbose argument but sadly got nothing useful out of it.
Here is the logfile from that:
[wasptool-pi4-timeout.log](https://github.com/daniel-thompson/wasp-os/files/4997392/wasptool-pi4-timeout.log)

As you can see it is still waiting for the prompt to show up as the timeout kicks in. That's why I wrote that it this could be about the bluetooth driver of the raspberry pi. In addition to that I wasn't able to flash "micropython.zip" from my pi either and as the error "No Notification received" sounds like a timeout thing too, it convinces me that there may be something wrong with it.

What do you think about that?

hrmckay, 2020-07-29:
I made a copy of the wasptool program and called it wasptoo2.  I changed line 35 to c.expect('>>> ', timeout=120) and ran sudo ./tools/wasptoo2 --rtc  I got the same error messages as before only this time it took two minutes for them to appear.  Again the watch was in an non operative state and I had to long press the button to reboot it.

It looks like the PI is sending a ^C and the watch is receiving it but the expected reply is not coming back to the Pi.

I connected a Bluetooth mouse to the Pi and it works perfectly so Bluetooth is operating.

Hugh

jeliebig, 2020-07-29:
But does the bluetooth mouse operate over BLE? Because that is how we connect to the watch, right? And I heard that this protocol is very different from normal bluetooth.

But again I have no clue. If bluetooth is not the problem (I really hope it isn't) then the firmware has to be, right?

hrmckay, 2020-07-30:
I assume the wasptool and other programs work fine on other Linux systems, but something about the Raspberry Pi is causing problems.  Unfortunately, my Pi4 is my only Linux system at the moment.

Hugh

daniel-thompson, 2020-07-30:
Certainly from what has been shared so far this looks like it is a RPi4 specific issue so I have updated the title accordingly.

If you have an Android device note that you can get access to the Python console using Serial Bluetooth Terminal. It is also likely there are also Windows tools that could be used to access the MicroPython console (the protocol is called the Nordic UART Service and it is is pretty common) although I have never tested anything like that.

hrmckay, 2020-07-31:
I am now confused.  Today I decided to try some debugging on this problem.  I booted my Pi4 and ran ./tools/wasptool --verbose --rtc  To my surprise it worked,  Below is the console log. I ran it again without --verbose.  It worked.  I ran it with and without sudo.  It worked every time.  This is the same Pi4, the same software (no updates applied) and the same PineTime as yesterday, when this command failed.   I decided to run the tools/ota-dfu/dfu.py command to upload micropython.zip.  It worked until it hit 86% and then it stalled and failed with the error reported previously.  I ran ./tools/wasptool --rtc and it failed.  The PineTime was not functioning.  I rebooted it and tried again.  Again --rtc failed and the PineTime locked up.  I thought maybe the failure of the dfu.py had caused a problem with the Bluetooth stack.  I exited my terminal window, opened another one and tried --rtc again.  It failed.  I rebooted the Pi4 and tried --rtc again.  It failed.  I powered off the Pi4, left it off for several minutes, then booted it and tried --rtc again.  It failed.

I do not understand what is happening.

Hugh
[wasptool_works.txt](https://github.com/daniel-thompson/wasp-os/files/5004350/wasptool_works.txt)

SteveAmor, 2020-07-31:
@hrmckay my Pi3B is a little flakey and requires reboots from time to time but I don't have anywhere near the problem you are experiencing with a Pi4 (I don't have a Pi4 to test).
I wrote a python app that uses the pynus files used by wasptool and the only way I could get it to work properly was to constantly switch the BT interface down and then up and also remove the cached BT devices. Here are the commands I used:
```
sudo hciconfig hci0 down
sudo hciconfig hci0 up
sudo bluetoothctl devices
# get each BT address from above and run the below to remove each address
sudo bluetoothctl remove <address>
```

jeliebig, 2020-07-31:
I just tried it with the script and now I can see the bluetooth symbol on my watch before it crashes but nothing else seems to happen.
Here is the log of what happened:
[wasptool-bt-restart.log](https://github.com/daniel-thompson/wasp-os/files/5006854/wasptool-bt-restart.log)

jeliebig, 2020-07-31:
I also have a raspberry pi 3 lying around and tested wasptool with it. 
I just cloned the repo on it and built submodules + softdevices. After running the command the first time, it was stuck at resolving services and the watch seemed fine. After that I deleted the device with bluetoothctl and restarted the bluetooth adapter. The next test resulted in the same timeout I received with the pi4. Watch crashed and initial prompt never came up.

Note: I use 32-bit raspi os on this pi3 and not the 64-bit os like I do on my pi4. So it seems like the problem does exist on both systems? And is not related to an os issue?

Here is the log again:
[raspi3-wasptool-timeout.log](https://github.com/daniel-thompson/wasp-os/files/5007786/raspi3-wasptool-timeout.log)

hrmckay, 2020-07-31:
I am using a 4Gig Raspberry Pi4 running the beta 64 bit Raspberry PI OS (I couldn't find a AArch32 Linux tarball for the GNU toolchain).  It is running BlueZ 5.50.  The Pi4 is the only PI supporting Bluetooth 5.  My PI is using WiFi for its networking, which uses the same chip as Bluetooth (interference?).  The PineTime is only a few feet from the PI.  I have a Pimoroni Fan Shim on my Pi to keep it cool.

Hugh

jeliebig, 2020-08-29:
Hello again,

I just want to report that the recent raspberry pi bluetooth update did not fix this issue. I still get the timeout and my watch crashes.
I was hoping that they fixed the bluetooth driver with that update but that seems to be not the case.

I hope we can find a solution to this somehow.

hakanrolsson, 2020-10-29:
Hi,
I have brute force way of getting this to work:
1) run commands suggested in comments above in a loop
`while true; do sudo hciconfig hci0 down; sudo hciconfig hci0 up; sudo bluetoothctl remove PineTime; tools/wasptool --verbose --rtc; done`
2) keep hitting ctrl-c if the pinetime disconnect from bluetooth
After a few tries it normally works.

jeliebig, 2020-10-30:
Thank you for that!
I will try it out and check if it works on my device too. If it does we can confirm that this is a bluetooth issue of the pis and not a issue with wasp-os, right?

daniel-thompson, 2020-11-11:
Please re-test with the latest master branch: https://github.com/daniel-thompson/wasp-os/actions/runs/358578049

The latest version has been upgrade from micropython-1.12 to micropython-1.13 and this includes bug fixes that reportedly improve BLE stability.

PS I would really like to thank the dev who informed me of the changes in the new micropython... but after ten minutes trawling through the bug tracker I still cannot find who it was. Nevertheless, many thanks are due for the heads up!

jeliebig, 2020-11-12:
YES! It finally worked with the latest master branch version! Thank you so much!

Couldn't get it too work with the "brute force" script from before as it just crashed my pinetime all the time. But I just flashed the new zip file and wasptool was able to set rtc instantly!

daniel-thompson, 2020-11-12:
Excellent. Thanks for the update.

Also I finally found who to credit! Thanks for @Carglglz for the pointer.

closed 2020-11-12 by daniel-thompson