wasp-os/wasp-os #47: Bricked P8 after following video tutorial

MirkoCovizzi, 2020-07-30:
Hi, I've followed [this](https://www.youtube.com/watch?v=VJoDtMy-4pk) tutorial and now my P8 is bricked. Display doesn't turn on anymore.

SteveAmor, 2020-07-31:
So, to check, you can at least get to the wasp-os bootloader splash screen?
If so after about 5 mins, does it go to the black screen?
If it does, there are many closed issues where people have had a black screen after flashing which should help you.

MirkoCovizzi, 2020-07-31:
Yes, I can get to the bootloader and yes, after some time it turns off and I would need to keep pressed the button for 5 seconds to reboot in the bootloader. I've checked the other issues and yes, it seems to be the same problem or something very similar at least. I will wait for the device to discharge completely as suggested by someone.

SteveAmor, 2020-07-31:
Cool. Are you able to console into the repl?

daniel-thompson, 2020-07-31:
Glad to see the edit... the original description worried me a little ;-). If you can still reboot back to the bootloader at will then there's still plenty of scope to explore without ripping the back off!

The device is showing up in DaFlasher as P8 which is great. That should mean you can get to the Python prompt to poke about. In the ideal case then `wasptool --console` (GNU/Linux systems only) should bring it up but you can also use tools such as Serial Bluetooth Terminal for Android to get the >>> that you need! 

If all else fails then waiting for the battery to run flat is worth a try... but IIRC any version of wasp-os that has P8 support shouldn't black-screen if there are problems with the flash then there should be a note about that on the display (and oddly, I didn't see those problems using DaFlasher to install on PineTime or P8).

Finally can you share the git SHA-1 for the version of wasp-os you have checked out.

daniel-thompson, 2020-07-31:
[micropython-v0.2-56-g70c1fb2.zip](https://github.com/daniel-thompson/wasp-os/files/5005659/micropython-v0.2-56-g70c1fb2.zip)

BTW, at this point "known working binaries" can be quite useful and, since we haven't had a release for P8 yet, I just built and tested the one attached here (built from https://github.com/daniel-thompson/wasp-os/tree/70c1fb25cabbebe7870a7e3e825a571b9c15b201 ).

Well, when I say "tested" I mean the clock shows up and the side button works... but I think that is enough for now.

MirkoCovizzi, 2020-07-31:
@daniel-thompson Hi, thanks for the rapid response. First of all, I've compiled from the commit `c3d9e3a0a2f8d983e6f2799a1dde5c17c7669ef3`. I've tried flashing the image you attached, but I'm getting an `Init SPINOR FAILED` on boot.

daniel-thompson, 2020-07-31:
Believe it or not that is good news since it is probably the stuck spinor problem (the one that I have never reproduced, not even on a later batch PineTime) and it will hopefully be fixed by letting the battery go flat!

MirkoCovizzi, 2020-07-31:
So, this happens because the flash is in a stale temporary condition? I might research into it when I have the time because it's interesting but also problematic. Anyway, as suggested, I'll let it go flat and then see what happens. The micropython I've compiled will probably boot by then.

daniel-thompson, 2020-07-31:
Pretty much. Several people found that immediately after replacing the stock firmware that wasp-os cannot read the ID from the flash. It is very odd behaviour (not least because I think the device is supposed to report an ID even when asleep).

On my TODO list is to attach a logic analyzer to a PineTime and see what commands the stock firmware sends to the spinor. I was hoping that would give us a clue!

daniel-thompson, 2020-07-31:
PS the source you used to build from should have worked ok, perhaps double check your compiler version

MirkoCovizzi, 2020-07-31:
The compiler I've used (checked with `arm-none-eabi-gcc -v`) is `gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1)` which seems to be the same version specified in the README.

MirkoCovizzi, 2020-07-31:
@daniel-thompson Can [this](https://github.com/atc1441/ATCwatch/blob/master/ATCwatch/flash.cpp) from Aaron help maybe?

MirkoCovizzi, 2020-07-31:
Update: after discharge I'm now welcome with the so awaited clock! I hope that this issue will help other P8 users if they'll ever encounter the same behavior until a stable fix has been found. So, for the time being, I think that it's a good idea to keep this issue opened.

daniel-thompson, 2020-08-01:
Isn't #43 sufficient for that? I'm not keen to keep multiple tickets open for the same issue... it simply fragments the discussion.

MirkoCovizzi, 2020-08-01:
Absolutely, but I think it would help in that case reformulating the title adding that also the P8 has the same problem because at first I thought it was only related to the PineTime when checking the issues some time ago.

closed 2020-08-01 by MirkoCovizzi

beriberikix, 2020-12-30:
I'm not my issue is related, so let me know if I should open a new issue.

I started my P8 with ATCwatch and DaFlasher. Everything worked fine, including custom firmware. I proceeded to build from source for the P8 without any issues. Using the DaFlasher I programmed following the [Installation Guide](https://wasp-os.readthedocs.io/en/latest/install.html#daflasher-for-android). It took the `bootloader-daflasher.zip` and I see the wasp-os bootloader screen. However, any operation now fails. I've tried `micropython.zip`, `FitBootloaderDFU2.0.1.zip`, ATCwatch, even `bootloader-daflasher.zip`, all fail. I also tried DFU from the nrf connect. It seems as if the bootloader loses connection before the first packet and reboots.

Any tips to troubleshoot? I have an android device with serial BT terminal and a macOS laptop.

miguelrochefort, 2020-12-30:
@beriberikix  I had the exact same problem yesterday when I tried to flash my Colmi P8 with wasp-os.

After 4-5 hours of trying everything I could think of and thinking I had bricked my watch, I repeated the last step with a different phone. It worked on the first try:

- Grab Huawei P20 (previous phone was Nokia 6.1)
- Copy "micropython.zip" to device
- Install DaFlasher
- Launch DaFlasher
- Select P8DFU
- Disable "Toggle Bluetooth" checkbox
- Select "micropython.zip"
- The progress bar slowly reached 100% (previously, it stopped between 0% and 11%)

I don't know if the problem was the phone or not, but if you have another one you should give it a try.

beriberikix, 2020-12-30:
Oh that worked! I wonder why it didn't on a Pixel 3a - but I'll take it. Thanks :)