wasp-os/wasp-os #49: Release micropython.zip and/or make documentation clearer

AlwaysLivid, 2020-07-31:
Although the steps (particularly the ones involving DaFlasher) mentioned in the readthedocs.io site seem clear, there's no explanation as to what `micropython.zip` is supposed to be and where to download it, despite the fact that other tools do have download links.

daniel-thompson, 2020-07-31:
Build instructions are here: https://wasp-os.readthedocs.io/en/latest/README.html#building-from-source

There is not yet a binary release that works with DaFlasher.

AlwaysLivid, 2020-07-31:
> Build instructions are here: https://wasp-os.readthedocs.io/en/latest/README.html#building-from-source
> 
> There is not yet a binary release that works with DaFlasher.

Understood. I've just been having a couple of issues with the compilers as of lately. Nevertheless, I still believe that it's  not immediately clear that the output of the building process is supposed to be a DFU-flashable ZIP-file called by the aforementioned name.

commit 66e8d3e5 (2020-07-31): docs: install: Move Building from source into the install guide

Direct links to the install guide mean some people are reading the docs
but never see the main docs index and, as a result, don't find the
instructions on how to build from source. Let's fix that!

Fixes: #49
closed 2020-07-31 by daniel-thompson