wasp-os/wasp-os #55: Bluetooth is unreliable and disconnects after a while

AlwaysLivid, 2020-08-04:
# Problematic behavior:
- Watch disconnects a short while after connecting it to my laptop (with a Qualcomm Atheros chip)
- Watch cannot connecting with two different Samsung phones via Bluetooth

However, it should be noted that the bootloader's DFU/Bluetooth functions work normally.

## Impact

- This behavior prohibits me from uploading, tweaking and testing applications written in Python by all means, meaning that I can only depend on uploading custom apps by compiling custom versions of `wasp-os` and updating via DFU. I'm not even sure if the process I just described is even possible, since it's undocumented and most probably not how you're supposed to do this stuff.

## Screenshot

Here's a screenshot of an upload being randomly interrupted by an unprompted disconnection.

[image]

MirkoCovizzi, 2020-08-04:
> meaning that I can only depend on uploading custom apps by compiling custom versions of `wasp-os` and updating via DFU. I'm not even sure if the process I just described is even possible, since it's undocumented and most probably not how you're supposed to do this stuff.

@AlwaysLivid It's certainly one way to do it. I'm doing it as well on my desktop since I don't have any way to create a Bluetooth connection with the watch. Is it the best solution? Absolutely not. Does it work? Yes, you can still do some stuff. To accelerate development for some things like UI you can use the simulator (through `make sim`).

AlwaysLivid, 2020-08-05:
@MirkoCovizzi I depend on GitHub Actions to compile things. So, it pretty much barely works for me.

daniel-thompson, 2020-08-09:
I suspect this will varies with different devices.

I am able to maintain a good connection to two different phones for an "unlimited" length of time (e.g. notifications work ;-) ) and my main laptop can connect robustly. However I do have another laptop that disconnects intermitantly.

AlwaysLivid, 2020-08-10:
> I suspect this will varies with different devices.
> 
> I am able to maintain a good connection to two different phones for an "unlimited" length of time (e.g. notifications work ;-) ) and my main laptop can connect robustly. However I do have another laptop that disconnects intermitantly.

Can you connect with your custom version of Gadgetbridge, the Bluetooth menu in Android itself, or both? Just wanting to be sure over here.

daniel-thompson, 2020-08-15:
> Can you connect with your custom version of Gadgetbridge, the Bluetooth menu in Android itself, or both? Just wanting to be sure over here.

You definitely can't pair from the Bluetooth menu. micropython (and therefore wasp-os) does not yet implement pairing. You should use either the modified gadgetbridge or Serial Bluetooth Terminal to test connection stability from Android devices.

AlwaysLivid, 2020-08-17:
Can confirm that the same occasional disconnections also do happen with Gadgetbridge. (It randomly disconnects after a while, but everything works normally while the device is connected.)

daniel-thompson, 2020-11-11:
Please re-test with the latest master branch: https://github.com/daniel-thompson/wasp-os/actions/runs/358578049

The latest version has been upgrade from micropython-1.12 to micropython-1.13 and this includes bug fixes that reportedly improve BLE stability.

PS I would really like to thank the dev who informed me of the changes in the new micropython... but after ten minutes trawling through the bug tracker I still cannot find who it was. Nevertheless, many thanks are due for the heads up!

closed 2020-11-27 by daniel-thompson