wasp-os/wasp-os #56: Gadgetbridge support

jcmb42, 2020-08-04:
Hi Daniel,

thanks for all you work! I really enjoy every update on this project and I'm eager on testing everything.
One question (and not exactly an issue): 
in a recent commit for the [initial notification support](https://github.com/daniel-thompson/wasp-os/commit/a01fb7df573e13a6f6b9fbf2e7688d4e9713df6d), you wrote:

> This requires a modified version of Gadgetbridge and currently works by
implementing the BangleJS protocol.
In Gadgetbridge ensure "Sync time" is *not* set and choose "Don't pair"
when adding the PineTime device.

Did you compile Gadgetbridge yourself? If so, what exactly is the modification you made, in order to get it to recognize Pinetime (P8 in my case).
I tried with the current version 0.45.1 release and had no luck so far. 

Thanks in advance! :)

AlwaysLivid, 2020-08-05:
[Check this out.](https://github.com/daniel-thompson/gadgetbridge/commit/732971d7e49d96df9073b121fe7f102f29cbd915).

daniel-thompson, 2020-08-09:
To get notifications working you will need the GadgetBridge fork that @AlwaysLivid pointed you at.

The other important tips are to go into settings and make sure ""Sync time" is **not**  set and choose "Don't pair" when adding the device.

jcmb42, 2020-08-10:
Thank you both very much!

closed 2020-08-10 by jcmb42