wasp-os/wasp-os #58: Port to TWatch 2020, ESP32

sactre, 2020-08-06:
Hello,
First of all, congratulations on this project.
I would like to make a port to work on the TWatch-2020, which is a watch based on the esp32 chip.
Screen: ST7789V
Axis Sensor: BMA423
RTC: PCF8563
Power Manager: AXP202
Touch Screen: FT6236U
Vibration
buzzer
IR
I see that you have the code prepared to accept different boards, but I think they all have to be based on Nrf, is it true?
On the other hand, any recommendation where to start?
Thank you very much.

daniel-thompson, 2020-08-09:
> I see that you have the code prepared to accept different boards, but I think they all have to be based on Nrf, is it true?

Currently all boards are based on nrf but I'd be happy to see other watch designs working too.

> On the other hand, any recommendation where to start?

wasp-bootloader and reloader will be of no use on this platform so you only have to worry about micropython and wasp-os itself.

From that point make a copy of wasp/boards/pinetime and hack the top-level Makefile to try and get an esp32 build going. First job is probably to write drivers/pcf8563! I think the main area you are likely to have problems with is the machine module (nrf is not well aligned to the other micropython ports so you will find some problems there).

sactre, 2020-08-10:
I'm working on the watch class, but when I use the ST7789 class you have, I get rotated 180º, and I don't see that you have a rotate function implemented. If I use the class https://gitlab.com/mooond/t-watch2020-esp32-with-micropython/-/blob/master/st7789my.py it looks fine.

daniel-thompson, 2020-08-13:
ST7789 can has a register, called MADCTL that changes the order the internal memory is accessed. Setting this to perform a horizontal and vertcal flip results in rotating the display by 180 degrees.

sactre, 2020-08-15:
Thanks, I already have the clock application working.
I am making a system compatible with wasp applications, but a little out of the way in philosophy. The Twatch has wifi and therefore internet, background tasks are important and as it has 8 MB of ram, I allow many tasks loaded in the background, each of them running an asyncio task.

closed 2020-10-13 by daniel-thompson

picpic020960, 2020-12-19:
Hello Sactre and Daniel
sorry for my bad english (French)
wasp firmware for Twatch 2020 ok on my watch.
How to use the simulator for some update in python firmware ( Timer , Alarm)
I tested a few things but now I'm lost
Thanks for help