wasp-os/wasp-os #60: Proposal: common bootloader for different firmwares

MirkoCovizzi, 2020-08-09:
Hi, I'm currently working on a custom firmware for the PineTime family of smartwatches.

There are different choices when it comes to bootloaders for the PineTime, like using MCUBoot as Lee Lup Yuen has extensively been researching https://github.com/lupyuen/pinetime-rust-mynewt. However his approach is a bit different than the one adopted in `wasp-os` because it assumes that the firmware is the one supporting GATT update and not the bootloader. From my point of view, I think it's a better choice to have a solid bootloader, like the one used in `wasp-os`, that is always present in case of problems thanks to the `watchdog` but also that is the one responsible for updates, so anybody can flash the firmware they want without bricking the device as long as the firmware complies with the partitioning separated by the 0x78000 address.

With that said, my proposal is to have a common bootloader that can be used by anybody that wants to develop a firmware for the PineTime family. I have already a repository that starts as a fork of https://github.com/daniel-thompson/wasp-bootloader which is https://github.com/OpenSmartwatch/common-bootloader.

This common bootloader is up to date with the `Adafruit` upstream and passes successfully the GitHub Actions builds.

The requirements for this common bootloader would be:

* Have an agreed upon common logo and have a way to customize it easily

* Build and install instructions should be general and not firmware specific (through `make` it's easy to integrate it into other existing build systems)

* Should be regularly updated from upstream

Finally, as you can see, I've created an open source organization called `OpenSmartwatch` with the aim of centralizing resources for the PineTime family since there is already a lot of fragmentation.
@daniel-thompson You are already using an Adafruit based bootloader and Aaron Christophel some time ago expressed interest into it as well so these projects could be the first firmwares released by the organization with the objective of easy interchangeability.

AlwaysLivid, 2020-08-10:
If you follow the conversations on the PineTime-related chatrooms, you'll quickly notice that a glued-up version of the PineTime is planned, which would be running on MCUBoot and the board would be inaccessible to SWD debuggers.

MirkoCovizzi, 2020-08-10:
@AlwaysLivid I'm following and I don't particularly like that direction. As I wrote above, the MCUBoot approach could be fine if it weren't so dependent on the fact that every firmware developer needs to implement its own BLE stack, even if it's already made like nimBLE. Afaik, with MCUBoot you have no way to connect to the bootloader through BLE so you lose the only way to recover a firmware if something bad happens. Yes, it has mechanisms to safely update or rollback a system, however it is not fail proof as long as firmware developers need to adapt their firmware with a BLE stack that supports GATT updates.

AlwaysLivid, 2020-08-10:
> @AlwaysLivid I'm following and I don't particularly like that direction, which is enforcing a specific bootloader choice with no possibility to change it for something else. As I wrote above, the MCUBoot approach could be fine if it weren't so dependent on the fact that every firmware developer needs to implement its own BLE stack with GATT update support, even if it's already pre-baked in something like nimBLE. Afaik, with MCUBoot you have no way to connect to the bootloader through BLE so you lose the only way to recover a firmware if something bad happens. Yes, it has mechanisms to update or rollback a system, however it is not fail proof.

The direction of the bootloader seems to be out of the scope of this conversation in this case scenario from my point of view. I'd definitely point to join the PineTime chat, as this is a topic that is currently being discussed and is obviously subject to change. Pragmatically speaking, support for the PineTime and probably a decent amount of other boards in the future could eventually fade away unnecessarily by sticking with one option and rendering it as a hard dependency from my point of view.

MirkoCovizzi, 2020-08-10:
@AlwaysLivid I'm already in the chat :)

AlwaysLivid, 2020-08-10:
> @AlwaysLivid I'm already in the chat :)

Oh, oops, sorry, I tried to communicate with you over the chat and failed, so I didn't think that you were in there. The point is that it's still something that's currently being actively talked about.

closed 2020-08-10 by MirkoCovizzi