wasp-os/wasp-os #62: Official communication channels

AlwaysLivid, 2020-08-17:
Hey there, I've been very interested in the project and I've been tinkering around with it over the past few weeks. However, I've found myself being a bit held back by the fact that official development-related communications are strictly restricted to GitHub's Issues section, rather than a chat platform (e.g. IRC, Matrix, etc.).

I understand that the introduction of such means of communication may elevate the amount of attention that has to be paid to this project, but I believe that it will increase productivity and improve the clarity between contributors as far as the direction of the project is concerned.

daniel-thompson, 2020-08-17:
Would be interested to hear some thoughts from @SteveAmor here. Steve is both a contributor and has been pretty active on the github issue tracker offering help to users.

On my side I agree that github isn't great for certain types of discusion... if anyone want so ask "would it be good to do X" or "it is better to do Y or Z" then the Issue tracker is a bit clumsy. More than that if someone wanted to post "look at the cool application I have written for wasp-os" then it simply doesn't fit at all!

However personally I don't think chat is an acceptable way to organise projects. Chat is an excellent lubricant and a useful secondary channel for a project but IMHO projects organised on chat alone tend to be insular. Decision making becomes opaque becauwe the chat is inaccessible and invisible to community members are who don't want or aren't able to keep the chat window all the time.

I also think we need to be careful about spliting the Pinetime community at this point by introducing wasp-os specific forums. It it true I have accepted (and worked on) ports to other devices but PineTime is still (proudly) the lead and recommended device for wasp-os. Therefore I'm be interested to know whether there is anything that can be  done to make forum.pine64.org a better home for discussion.

PS I have created a IRC channel (#wasp-os on freenode.net) but I haven't decided whether to recomment it in the docs.

AlwaysLivid, 2020-08-17:
We could make use of alternative solutions in order to prepare roadmaps. I believe that chatting is an acceptable means of deciding on the details on the spot, without "too much e-mail". That could also solve the problem of "invisible decision making", or, in other words, the potential lack of transparency. A different approach that countless different projects take is the introduction of logging inside of the channels themselves. I'm personally uncomfortable with them, but I'm (admittedly) definitely in the minority.

I don't believe that any sort of subforum will partition the PineTime community further, considering that no project can obviously go fully independent at this stage. (e.g. Discussions regarding restrictive licenses surrounding certain parts of the PineTime, or bootloader-related stuff, which is a topic that particularly concerns wasp-os for the most part, as well as the ATCwatch firmware. I'd imagine that there's going to be a conflict again which would also require mutual resolution and discussions once the Gadgetbridge stuff progresses on multiple different firmwares with different capabilities.)

daniel-thompson, 2020-11-08:
I'm going to put this on ice for now. I think the wasp-os community is currently too small for chat channels to effectively, esepcially so given that I don't really "do chat" except as a social lubricant for more durable comms channels.

closed 2020-11-08 by daniel-thompson