wasp-os/wasp-os #64: gadgetbridge on PineTime no longer responding

graynada, 2020-08-18:
I have been working on a modified clock for the wasp-os on the PineTime and gadgetbridge.py was performing as expected but has suddenly stopped working.  I have the modified Gadgetbridge app install on my Android device and I had the 'find' function and 'notify' commands on the PineTime responding to outputs but now there appears to be no response to any output from the Gadgetbridge app.

I get a connection but the GB(cmd) function of gadgetbridge.py does not seem to be being called in response to an output from the phone.

I am pretty sure it is not the Android end as I have cleared all the storage of the app, uninstalled, and reinstalled with no change but I have also rebuilt a new wasp-os from a fresh clone of the master github and no change.

It is probably something stupid I have done but I am going round in circles trying to resolve.  Is anyone else having issues?

AlwaysLivid, 2020-08-19:
Not sure if your PineTime is frozen or anything like that. I'd definitely try deleting Gadgetbridge's cache and data though.

graynada, 2020-08-19:
Hi, thanks for your reply.

I probably didn't explain it very well but yes I have clear the Gadgetbridge Android app cache and data and uninstalled and reinstalled the app completely with no change.  The Gadgetbridge app detects the PineTime and happily connects (not paired, sync time deselected) but commands appear to no longer be received, or at least passed, to the gadgetbridge.py GB(cmd) function.

My PineTime is not frozen as I can reflash successfully from my laptop and via DaFlasher and I have tired 2 versions of wasp-os, the master and my modified version, which both perform as expected except for the reaction to Gadgetbridge outputs.