wasp-os/wasp-os #65: Swiping is detected in sleep state

Siroj42, 2020-08-19:
When the watch is in a sleep state, it will somehow record if the user swipes over the display. This can be distracting because as soon as the device is activated again, the action the clock app associates with the swipe will execute. Therefore, if someone wants to see the current time, but accidentally swiped over the touchscreen, he will not be able to see the time. Only the latest directional swipe is cached in this way. I only have a PineTime, so I am not sure if the issue also exists on other devices.

SteveAmor, 2020-08-20:
I have observed this on a P8 watch with the CST816S touch sensor (which is the same sensor as the PineTime). It doesn't occur on a newer P8 with the 716 touch sensor. 
I believe the fix for this would be to put the touch controller to sleep when the watch is asleep. This will also increase the battery life :-)

daniel-thompson, 2020-08-20:
Er... I think the touch controller is already asleep when we sleep the watch, there's certainly code that purports to do that (https://github.com/daniel-thompson/wasp-os/blob/master/wasp/drivers/cst816s.py#L94 ) although I haven't done the power measurements.

Either way we already toggle the reset pin during a wake up so it is odd that an event survives the reset... perhaps we'll just have check for events just after reset and discard them.

graynada, 2020-09-03:
I tried adding a watch.touch.rest_touch_data() to wake() in wasp.py and that seems to do the trick for me.

```
def wake(self):
        """Return to a running state.
        """
        watch.display.poweron()
        if 'wake' in dir(self.app):
            self.app.wake()
        watch.backlight.set(self._brightness)
        watch.touch.reset_touch_data()
        watch.touch.wake()

        self.keep_awake()
```

commit 6bf350a6 (2020-09-21): drivers: cst816s: Clear the event buffer during a wake up

Currently with CST816S controllers (but not CST716S controllers) then a
swipe delivered whilst the device is asleep will sometimes be processed
after we wake it up. That's never likely to be useful. Fix this by
explicitly clearing the event buffer as part of the wakeup sequence.

Reported-by: Siroj42 <siroj42@users.noreply.github.com>
Fixes: #65
closed 2020-09-21 by daniel-thompson