wasp-os/wasp-os #68: Control display brightness through PWM (P8, PineTime)

MirkoCovizzi, 2020-08-24:
Hi!

This is just a simple proposal of a feature I'm experimenting that works wonderfully. I'm not making a PR because it would require a change in the Settings UI by allowing up to 100 levels of brightness. In any case, here is the code to enable PWM control, in a transparent way, on wasp-os.

```
class Backlight(object):
    hi = Pin("BL_HI", Pin.OUT)
    pwm0 = PWM(0, pin=hi, freq=PWM.FREQ_16MHZ, duty=80, period=160000)
    pwm0.init()

    def __init__(self, level=1):
        self.set(level)

    def set(self, level):
        self.pwm0.deinit()
        if level <= 0:
            self.hi(1)
        else:
            d = 100 - level
            self.pwm0 = PWM(0, pin=Pin("BL_HI", Pin.OUT), freq=PWM.FREQ_16MHZ, duty=d, period=160000)
            self.pwm0.init()
```

miguelrochefort, 2021-01-03:
Does it allow brightness to go lower than _low_, or does it only add sub-levels between _low_ and _high_?

thiswillbeyourgithub, 2023-01-15:
replying to @miguelrochefort , yes it allows to reduce the minimum brightness. I'm working on a PR to enable this.

Thanks a lot to @fgaz  in #373