wasp-os/wasp-os #69: Stuck on bootloader (Colmi P8)

tomdoughty62, 2020-08-28:
Similar to this issue - https://github.com/daniel-thompson/wasp-os/issues/41
I'm using a Colmi P8.
Having used Daflasher and the nRF Connect app on Android, I've flashed over the daflasher bootloader, but after flashing over the Micropython.zip file the watch remains in the bootloader mode.
When I press the button once, it shows the wasp-os logo for a couple of seconds, then one for one quick second before going to a black screen. When I attempt to flash over the bootloader again the watch just restarts and displays the wasp-os logo again, and uploading different versions of the micropython.zip (from older commits) the results are sadly the same.
Any assistance would be greatly appreciated.

hrmckay, 2020-08-29:
When the watch is in the black screen have you tried issuing commands to it with the Serial Bluetooth Terminal app on Android?  When my Pinetime booted to a black screen it was because main.py had failed.  But when I issued the commands 'import wasp' and 'wasp.system.run()' (without the quotes) the watch came to life.

Hugh

tomdoughty62, 2020-08-29:
Oh thanks that's great advice! After importing and running that command I got:
AttributeError: 'module' object has no attribute 'system'

So I can connect and tinker with my P8, but wasp may not be working correctly. 

But at least I can connect to it and send code

daniel-thompson, 2020-08-29:
Try the same commands after booting into safe mode (hold down the physical button thru the bootloader and until Init button appears on the screen, then release). That will skip some  initialization that allows you to see what is really happening during the `import wasp` (results of module import are cached and I think it may have previously been imported).

tomdoughty62, 2020-08-29:
Its just reported an invalid sensor. I've attached a screenshot. 
[image]

daniel-thompson, 2020-08-29:
So, I think there is good news and bad news here.

The good news is that I suspect if you comment out a single line then the watch will start booting OK:
https://github.com/daniel-thompson/wasp-os/blob/01ac0d93c30a219e5cd2aa47551d1730fd34c627/wasp/wasp.py#L119

The bad news is that this will disable the step counting feature!

Basically I suspect the step counter in your watch is a different type to the one that others have (e.g. it has been replaced with an alternative for supply chain or cost reduction reasons). Happily with the watch working you will be in a good position to try both use the rest of the watch **and** understand more about the step counter. A good first goal will be to see if the sensor is very different or only a sublte revision... a quick way to do that is to enable the BMA42x driver debug mode. From memory (e.g. untested but roughly the right idea) would need something like:

    watch.accel.debug(True)
    watch.accel.reset()

daniel-thompson, 2020-08-29:
Looks like I made a mistake in the instructions, the first line should have been `watch.accel._dev.debug(True)` (the second line is correct as it was).

tomdoughty62, 2020-08-30:
Hello,

Thank you very much for all the help. The watch switched on as soon as I'd commented out the StepCounterApp. Absolutely chuffed to bits that I can now start tinkering.

When I used watch.accel_dev.debug(True) and accel.reset() I got the following output:
BMA42x read: 24 00 [00]
Traceback (most recent call last):
   File "<stdin> in <module>
   File "drivers/bma421.py", in reset
   ValueError: invalid sensor

Also lovely to see more documentation on this. This project is very accessible for newbies like me.

MirkoCovizzi, 2020-09-06:
The new accelerometer is the SC7A20. I have just received 2 new P8 smartwatches and both have it. The SC7A20 doesn't have hardware step counter, so it's going to be needed a step counter algorithm implemented by software.

tomdoughty62, 2020-09-12:
@MirkoCovizzi How would I discover that now that I've got Waspos installed?

MirkoCovizzi, 2020-09-12:
@tomdoughty62 You need to either open it up and see for yourself or trust other people that did that before you. At the moment there are at least 3 variants of the P8, the main perceivable difference is the accelerometer and there are 2 possible models, so if the "regular" wasp-os doesn't work without disabling accel, then it means that it has the other model.

daniel-thompson, 2020-09-17:
This issue (not booting) should be fixed by 889115f4b081428ebfcf0f71c7ab405fb22c2527 ("wasp: Allow app initialization to fail") and this is now in the master branch. Could someone with a P8a test and confirm?

PS I have also updated the docs to explain what is happening so providing the changes work I'll probably close this ticket...SC7A20 support would be nice but at present its status is "pull requests welcome":
https://wasp-os.readthedocs.io/en/latest/install.html#colmi-p8

daniel-thompson, 2020-11-08:
Haven't seen a confirmation... but it's been nearly two months so I'm going to assume this is fixed.

closed 2020-11-08 by daniel-thompson