wasp-os/wasp-os #70: Boot loop

balkian, 2020-09-12:
I've run into a boot loop while trying to flash a device for the first time using the installation instructions.

The bootloader part using DaFlasher went great, but after that the watch reboots at the beginning of every new upload (e.g., `micropython.zip` and `DS-D6-adafruit-back-to-desay-sd132v201.zip`).

Using `wasptool` to upload `micropython.zip` works, but then it gets stuck in a bootloop, until you get into safe mode by holding down the button.

I've rebuilt the project several times and tried removing `boot.py` from the manifest, to no avail. The console does not work either, I've tried with Serial Bluetooth Terminal (`no serial profile found`) and wasptool (`connection timeout`). 

At this point I don't know what to do, so any help would be appreciated.

The device is this one: https://www.aliexpress.com/item/4000768483197.html

Last but not least, thank you for this great project. It was the reason I bought the watch in the first place :)

SteveAmor, 2020-09-14:
Hi @balkian you could try two things. First, check this fix out https://github.com/daniel-thompson/wasp-os/issues/69#issuecomment-683284232 and second, wait for the battery to go fully flat so everything resets.  
Could you confirm if you have the magnetic charger. If so, it will probably have the new accelerometer chip so my first thing to check may work.

balkian, 2020-09-14:
Thank you, @SteveAmor 

Indeed, it comes with the magnetic charger. I've tried both solutions but it's still looping. I guess my issue is different from #69, because I can't connect to the console even after commenting the manifest.

daniel-thompson, 2020-09-16:
I'm afraid so say that most of the reports I've had recently about boot loops have not actually exhibited any looping (unless the button is held down when looping is expected).

So... please hold down the button until the boot logo appears then describe what the device shows you and what you do it! You might also find the Troubleshooting guide useful since you can check if the device will correctly enter OTA mode, Safe mode and normal mode.

Given you are having problems with Serial Bluetooth Console it would also be good to know what name the device advertises in each of the modes.

daniel-thompson, 2020-09-16:
Oops... troubleshooting guide is here: https://wasp-os.readthedocs.io/en/latest/install.html#troubleshooting

daniel-thompson, 2020-10-26:
It's getting close to six weeks without an update... I'm going to close this for now. Feel free to follow up when you have got something for us to analyze.

closed 2020-10-26 by daniel-thompson

balkian, 2020-11-17:
Thanks a lot, @daniel-thompson. And I'm sorry it took me so long to reply. The latest commit  fd0e0d77aa9ca90f0cb110a31fb098d189b87114 is working fine.

It took several attempts and I used different methods, but for future reference I will try to summarize it. The device showed as `P8DFU`, IIRC. It would boot into safe mode after holding the button, but when I tried to upload through DaFlasher or nRF Connect it just failed after a few seconds.

Following your advice, I started from scrach on master and built from source. Eventually I managed to upload the bootloader using DaFlasher, then I uploaded micropython using `dfu`:

```
tools/ota-dfu/dfu.py -z build-p8/micropython.zip -a <MAC> --legacy
```
After that, the watch booted `main.py` as expected :)

Btw, I'm on arch, so to build from source I had to install `arm-none-eabi-gcc` and add it to PATH:

```
export PATH=/opt/ccstudio/ccs/tools/compiler/gcc-arm-none-eabi-10-2020-q2-preview/bin/:$PATH
```