wasp-os/wasp-os #73: Setting rtc

tht-jxny, 2020-09-27:
Hey there. I want to set the correct time and date on my PineTime and tried several approaches but unfortunately, nothing has worked so far. I would appreciate it, if the documentation would explain several other ways to set the rtc. 
These are the methods I tried:
1. Using `tools/wasptool --rtc` produces (in normal mode and bootloader): 
``` Shellscript
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/wasptool", line 285, in <module>
    console.expect('Exit console using Ctrl-X')
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 341, in expect
    timeout, searchwindowsize, async_)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0xb61c4990>
command: tools/pynus/pynus.py
args: [b'tools/pynus/pynus.py']
buffer (last 100 chars): ''
before (last 100 chars): 'ecurity policy blocked the reply, the reply timeout expired, or the network connection was broken.\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 2550
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('Exit console using Ctrl-X')

```
2. Using `tools/wasptool --console` fails as well with the same error message
3. `Serial Bluetooth Terminal` is unable to connect to the PineTime
4. I thought there was a way to set the rtc in the payload or in the source code but I was unable to remember how.
5. Connecting my Phone to the PineTime over normal Bluetooth Settings results in *Pairing rejected by PineTIme* (As expected)

Are there any other ways to set the rtc?
I wondered if I should install a modified Gadgetbridge as mentioned in #56 
Is there a way to set the rtc with nrfConnect?
Thanks for your answers.

BTW: I had some trouble flashing the bootloader to a totally empty PineTime (mass erase) with my Raspberry Pi. Should I complement the documentation with my working procedure?
@daniel-thompson Do you want additional applications in this repo to make the SmartWatch more appealing? I'd love to share Games and some useful applications such as a Calculator. Nevertheless, I would fully understand if you desire a different development in your repository (e.g. presentin only the base so others can personalize their watches)

hrmckay, 2020-09-28:
What system are you running the wasptool commands from? If it is a Raspberry Pi then I and others have had problems with these commands always failing if sent from a Pi.  Others say the commands work from other Linux systems.  Serial Bluetooth Terminal works for me however.  What is on the screen of the Pinetime when you try these commands?  The screen with the (wrong) date and time?  What is the error message you get from Serial Bluetooth Terminal?

Hugh

tht-jxny, 2020-09-29:
I indeed ran these commands from my Raspberry Pi. This is because connecting my Laptop to the watch produces even more errors. When I run these commands the watch shows the wrong date and time so I suspect it's "normal mode". Serial Bluetooth Terminal doesn't list the PineTime on the "Scanning..." page under "Devices". Could you please tell me again in which state my watch should be whilst connecting? Normal mode?  Bootloader? Safe mode? 
Thanks a lot @hrmckay !

daniel-thompson, 2020-09-29:
For setting the time then normal mode is fine. The watch should appear (as "PineTime") in Serial Bluetooth Terminal when you do a Bluetooth LE scan (assuming your phone has BT LE support). When you click on "PineTime" it should connect.

tht-jxny, 2020-09-29:
In my phones normal Settings, PineTime does appear. But the list in Serial Bluetooth Terminal is completelty empty... I assume it doesn't matter if the display is on or off. Am I correct?
Tomorrow, I could try another phone.

Any ideas if reflashing the firmware could solve the problem (Connection crashes during firmware upload in the nrfConnect-App but flashing works fine with wasptool on my Raspberry Pi)

What about the other methods I tried?

daniel-thompson, 2020-09-29:
Display state doesn't matter since watch bluetooth is active all the time. It be suprised if reflashing will help. I've never known a device that boots to the clock that is not fully working. Basically if you want download the firmware over the air then Bluetooth code is working and if the clock comes up then rest of the watch is working OK.

In both cases the debug needed is probably on the other device (the error messages you see in wasptool basically mean "I can't find a wasp-os device").

daniel-thompson, 2020-09-29:
Oh yes... and more apps are very welcome. I'm perfectly happy to collect additional apps in the wasp-os repo although for memory saving reasons things like games (including my own game of life program) will probably be disabled by default.

Currently the memory that needs saving is RAM rather than FLASH so we could include them by default but require changes to main.py for the user to activate the apps they are interested in.

tht-jxny, 2020-09-29:
To be clear, the watch is fully functional. It just shows let's say 10 March 2020. 

I assume that uploading the firmware with wasptool works because then I have to add the MAC address. Is there an option that one can add the MAC address to every wastool command?

Concerning additional apps, I propose I open a new issue where I first upload some screenshots and maybe also the code, so that there is a somewhat similar design language throughout wasp-os and that possible flaws in the code can be corrected.

hrmckay, 2020-09-30:
The problems other people were having with the Raspberry Pi are documented in #46.  

You said the scan list in Serial Bluetooth Terminal is completely empty.  Even after you hit the Scan button?  On my phone it always shows at least 6 unknown devices in addition to the Pinetime.  Maybe these are other Bluetooth devices I have or my neighbours have.

tht-jxny, 2020-10-01:
This is what I'm seeing in the settings and in Serial Bluetooth Terminal:
[image]
[image]

daniel-thompson, 2020-10-01:
This looks a lot like a problem on the phone. Can you see (and connect to) the PineTime using a BLE diagnostics app such as nRF Connect?

For new apps I'd recommend using the pull request approach since this is the best way to post code for review. BTW as you'll see from the settings app the design language mostly comes from Material Design.

tht-jxny, 2020-10-06:
I am able to connect to the PineTime using nRF Connect. Is there an option to set the RTC in nRF Connect? Do you have other options for me to set the rtc? Are some in the making?

In my settings app, I included also an option to set the time before the watch goes to sleep to 7, 15 or 30 second. Is this something for your repo as well?

daniel-thompson, 2020-10-10:
> I am able to connect to the PineTime using nRF Connect. Is there an option to set
> the RTC in nRF Connect?

Sort of... but that's not why I asked. I just wondered if if would offer any clues about why Bluetooth Serial Terminal cannot enumerate anything when running on your phone. When you connect to a device with nRF Connect you can examine the services that are advertized and check that they make sense.

There are other Android tools chat can speak the same protocol. nRF Toolbox for BLE but its not really very easy to use for UART stuff (strange newline handling and each packet from the watch is printed seperately).

> Do you have other options for me to set the rtc? Are some in the making?

Eventually I'd like to extend GadgetBridge to set the time but I'm not working on that at the moment.

I think it is a big loss not to have a working console since most of the benefit of having an interpretter built into the watch are lost when you are unable to communicate with it. In that context I view Android as the "other option" and it is still a poor choice compared to wasptool since things like `wasptool --exec` and `wasptool --binary --upload` are really cool!

On the other hand... you've written some neat applications despite not having wasptool working so there's definitely room for more than just my opinion about how to best use wasp-os. It is certainly possible to write a time setting app (there's not currently any widgets to help but you could do something quick and dirty based on the Notification Test UI in the self-test program).

> In my settings app, I included also an option to set the time before the watch goes to sleep to 7, 15 or 30 second. Is this something for your repo as well?

I'm slightly reluctant. My attitude to settings is strongly influenced by writtings such are: https://ometer.com/preferences.html .

Additionally with a "companion" device such as PineTime (e.g. something that needs BLE tether to for many of the interesting cases) there is also the question of whether a setting should be controlable from the watch or from companion app. Broken record I'm afraid I think the changing the display timeout can a wasptool one-liner.

PS I did get a RPi4 recently and will take a look at some point (RPi being an cheap way to get a "real" GNU/Linux system). Don't be too optimistic though... there's no particular reason to expect me succeed where others haven't (but I will take a look anyway).

hrmckay, 2020-10-17:
Another way to set the rtc is using the android app Serial Bluetooth Terminal.  While the watch is running wasp-os connect to the watch with the app and send a ^C.  You have to edit a macro and have it send hex 03 to send a ^C.  After the ^C you can send the commands (without the quotes) 'watch.rtc.set_localtime((yyyy,mm,dd,HH,MM,SS))' and then 'wasp.system.run()'.

Hugh

tht-jxny, 2020-10-17:
@daniel-thompson 
> It is certainly possible to write a time setting app (there's not currently any widgets to help but you could do something quick and dirty based on the Notification Test UI in the self-test program).
>>
This is what I've done now. I use my calculator app I wrote some time ago. When somebody types in the localtime-tuple, the time is set. My only problem is: The function `set_localtime` cannot be executed.
`dir(watch.rtc)` brings up:
``` 
['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_epoch', '_lasttime', 'get_localtime', 'get_time', 'get_uptime_ms', 'update', 'uptime']

```

Any ideas?

daniel-thompson, 2020-11-08:
No ideas here I'm afraid.

I worked with another user to use set_localtime() via Serial Bluetooth Terminal and didn't see any problems.

tht-jxny, 2020-11-15:
So I now tried all my devices that have a Bluetooth module and finally Serial Bluetooth Terminal worked on one of them. I now have a watch that can display the correct date and time ^^. I'd call this a partial success. 
Nonetheless, I'd like to find out what other possibilities there are to set the rtc and what the above mentioned error means. I cannot imagine how this function is not callable. Anyways, I will now close this issue as it is unlikely that this problem has something to do with the watch or the firmware, but rather with my devices.

closed 2020-11-15 by tht-jxny