wasp-os/wasp-os #77: Why don't you use LVGL as graphics library?

seniorivn, 2020-10-13:
https://docs.lvgl.io/latest/en/html/get-started/micropython.html#so-what-does-it-look-like

just ordered p8, plan to use some custom os with it, thought micropython with lvgl would be the best option.

daniel-thompson, 2020-10-13:
wasp-os currently does not enable lvgl support because it looks like it would need too much flash space (something like an eighth of the total). At present I don't think there is enough spare space to allow it.

You are welcome to try... but I suspect your first task would have to be to free up space by porting micropython/nrf to a smaller bluetooth stack (currently it uses SoftDevice S132 which has more features than we are currently using).

seniorivn, 2020-10-13:
@daniel-thompson got it
what about https://lupyuen.github.io/pinetime-rust-mynewt/articles/micropython#nimble-bluetooth-stack this proposal and or its bluetooth stack?

daniel-thompson, 2020-10-26:
> what about https://lupyuen.github.io/pinetime-rust-mynewt/articles/micropython#nimble-bluetooth-stack this proposal and or its bluetooth stack?

It's a pretty cool hack... on the other hand last time I looked at the code if felt more like a proof-of-concept than a solid base for engineering. More specifically the port is based on modifying ports/nrf until it comes alive with mynewt rather than implementing a proper micropython port for mynewt (port/mynewt).

closed 2020-11-06 by daniel-thompson