wasp-os/wasp-os #79: Bluetooth sockets?

mel4tonin4, 2020-10-14:
Hello.

This is a question, more than an issue. 

Does _WASP-OS_ allow bluetooth communication between an application on the watch and other devices?

The purpose could be, for example, creating a "companion application" running on a cheap smartwatch, for people to test it and eventually purchasing a more costly watch. So this issue is related to ["frontend/backend model for wasp-os apps"](https://github.com/daniel-thompson/wasp-os/issues/78).

Is this at all possible with the kind of devices on which _WASP-OS_ is currently running?

Thanks!

seniorivn, 2020-10-14:
@mel4tonin4 right now there is [a limited gadgetbridge android app support](https://github.com/daniel-thompson/wasp-os/blob/13d9095872ac59b9e466fb6051aa79e4557b5afa/TODO.rst#L77)
all supported devices have bluetooth (ble) and depend on it to flash the os

mel4tonin4, 2020-10-16:
@seniorivn Thanks for the answer.

_Gadgetbridge_ would not be enough for what I had in mind.

What should I learn to understand the amount of work needed to use bluetooth sockets in a _WASP-OS_ app, modifying _WASP-OS_ itself, if necessary?

mel4tonin4, 2020-10-16:
I saw that the _Bluetooth_ implementation is part of _micropython_ and not of _WASP-OS_, so I have to read about that. Any suggestions are appreciated, though.

daniel-thompson, 2020-10-16:
> I saw that the _Bluetooth_ implementation is part of _micropython_ and not of _WASP-OS_, so I have to read about that. Any suggestions are appreciated, though.

I don't understand what you mean here. wasp-os has a richer set of Micropython features enabled that the nrf defaults and I don't recall anything being disabled. We don't guarantee that will never change but, right now, if it works in the Micropython nrf port it should also work in wasp-os .

mel4tonin4, 2020-10-16:
@daniel-thompson 

> I don't understand what you mean here.

What I mean is just that I just looked for a bluetooth module in the _WASP-OS_ sources and could not find anything. Then I stumbled upon the `ubluetooth` _Micropython_ module: https://docs.micropython.org/en/latest/library/ubluetooth.html .

That was unexpected. I would think hardware support as a part of the OS and not of the language in which the OS is written. But I'm completely unfamiliar with this kind of devices, their firmware and _Micropython_. It's the first time I have a look into a similar system.

I didn't mean anything more than that. In particular, I didn't mean that the bluetooth feature is not available in _WASP-OS_, just that is not provided by _WASP-OS_, strictly speaking, but by _Micropython_.

Sorry for the misunderstanding.

daniel-thompson, 2020-10-16:
No worries.

OS means something pretty different when used in RTOS and when used in Unix and Linux situations. The use here is perhaps closer to the meaning used by Linux distros. Overall I think it is reasonable to describe an application runtime environment as an "OS" (even if most of the runtime library is inherited from uPy) but it does depend from what angle you look at things.

To be honest the "os" in wasp-os serves several purposes which are not technical anyway: compared to just "wasp" it is easy to find with a search engine and having a post-fix makes it much less likely to have problems with trademarks.

Moving away from philosophy, take a look at #81 looks like an interesting to use the BLE API (I haven't looked at it in detail yet so there will probably be review comments but regardless of whether it gets pulled it is an interesting use of the API).comments...

mel4tonin4, 2020-10-17:
@daniel-thompson  Thanks for the explanation!

#81 is something. As far as I can understand, ubluetooth does not implement _RFCOMM_ (a.k.a. "Bluetooth sockets" on _Android_), so I would need to use _GATT_, like they do in #81. Otherwise I would have to implement _RFCOMM_ myself.

Thanks again.

daniel-thompson, 2020-10-17:
I could be wrong but I understood RFCOMM to only be available as part of Bluetooth Classic and that there is no standardized equivalent for Bluetooth Low Energy. AFAIK for BLE the closest thing is the Nordic UART Protocol is used

mel4tonin4, 2020-10-24:
Thanks, @daniel-thompson. I'm going to read about the various protocols that are part of BLE, so that I can make a better evaluation of the effort required to implement my application.

closed 2020-10-26 by daniel-thompson