wasp-os/wasp-os #82: Gatttool and hcitool depreciated from arch

Izaic, 2020-10-16:
Gatttool and hcitool have been dropped from arch's bluez-utils package for a long time.

Would it be possible to replace the dependency? Is there even any distros with these tools packaged still, or do you have to manually build them?

daniel-thompson, 2020-10-16:
This is an inherited dependency.

If you just want to raise a ticket then please add it against the correct component (either https://github.com/dingari/ota-dfu-python or https://github.com/daniel-thompson/ota-dfu-python ). That is also the right place to send pull requests.

Alternatively if you aware an an alternative dfu implementation that wasp-os could use instead then please let me know.

closed 2020-10-16 by daniel-thompson

fgaz, 2021-09-03:
Your fork doesn't have issues enabled, is there another place where I can open a ticket?