wasp-os/wasp-os #83: Is there any support for sound / alarm / chime on the P8?

github-dr, 2020-10-19:
Firstly, thank you for wasp-os, I've got it working successfully on a P8 and now started customizing it to my use case.  It's exciting to have a fully controllable canvas on my wrist.  One thing I haven't found in wasp-os is any API for controlling an alarm/sound on the watch, which is making me wonder if the device actually has a sound chip?  I'm currently using the vibrate function for my alarm use case, but the vibration isn't very strong and it's easy to miss if you aren't wearing the watch strapped tightly.  An audible beep / chime would be better if possible.  Do you know if there is a speaker in the watch?  I flashed the original watch immediately after it arrived so I never got a chance see if the original supplied firmware had a beep/alarm/sound function.  The https://www.colmi.com/products/p8-pro-smartwatch page mentions an "alarm", but it is unclear if that's just the vibrate function for if there's some sound chip in the device.

SteveAmor, 2020-10-20:
Hi @github-dr, there isn't a sound chip or speaker in a P8 so the alarm mentioned must be the vibrator.

github-dr, 2020-10-23:
Good to know, thanks for confirming Steve.

closed 2020-10-23 by github-dr

thiswillbeyourgithub, 2022-02-25:
Incidently, I discovered that `wasp.watch.vibrator.pulse(95, 500)` actually produces a decently strong high pitch sound. This could definitely be used as a chime.

@daniel-thompson do you think this kind of vibration is bad for the watch?

Test it yourself : `for a in range(3) :wasp.watch.vibrator.pulse(95, 500) ; time.sleep(0.1)`