wasp-os/wasp-os #84: Trouble using "make" term

SC-Teuton, 2020-10-20:
I am attempting to install wasp-os however, each step I arrive at that involves the 'make' term flag up this error and I am unsure as to what I have done incorrectly:
[image]

Help would be much appreciated as I look forward to using wasp-os!

SteveAmor, 2020-10-21:
Hi @SC-Teuton, the install instructions are for Linux and make is not available in Windows. However if you have Windows 10 with WSL, you can follow the instructions here:
https://stackoverflow.com/a/54133058

SC-Teuton, 2020-10-21:
I apologise to come back with another question with another probably very simple answer.
I have followed the steps to install wasp-os as have been linked above, I installed the ARM Toolchain through the answer in this link: https://unix.stackexchange.com/a/377350 since I'm on windows 64-bit and the website linked in the installation guide only supports 32-bit.

Everything, up to this point, has seemingly worked well or has been fixed by research. The main problem I am facing now is this with the 'make sim' command. I get the following output: 
[image]

I do see the 'XDG_RUNTIME_DIR not set in environment' error however havve done a large amount of searching that yielded no working result.
How may I fix this?
Again, apologies for the questions that are most likely easy to fix.

hrmckay, 2020-10-22:
WSL on Windows 10 does not support graphics.  Microsoft says it is coming "soon" but it is not available now.  The wasp-os simulator opens a graphics window and WSL cannot handle that.  I found that I had to run Debian Linux under an emulator in order to get the simulator to work.

Hugh

SC-Teuton, 2020-10-22:
> WSL on Windows 10 does not support graphics. Microsoft says it is coming "soon" but it is not available now. The wasp-os simulator opens a graphics window and WSL cannot handle that. I found that I had to run Debian Linux under an emulator in order to get the simulator to work.
> 
> Hugh

Hi, Hugh.

Thank you for the response, though that is a shame to hear.
Are there any resources out there that you can offer on running a Debian Emulator, please?

And I assume that means it's still possible to actually compile and flash code to the PineTime (using the Android app or cables in the dev kit) despite this? I have mine currently being shipped to me so I'm unable to test it on my own watch just yet.

Cheers~

hrmckay, 2020-10-23:
I run Debian under an emulator called Parallels on a MacBook Pro.  For a PC running Windows there are several emulators, QEMU and VirtualBox are two of the better known ones.

From your previous screen shot it looks like only the "make sim" command failed.  The other commands would have created the necessary files for the watch, like micropython.zip.  I copy these files to my Android phone using DropBox and then use the Nordic nRF Connect app to send the files to the watch.  As I understand it the current production PineTime, not the dev kit, has a bootloader incompatible with wasp-os but Daniel has instructions on fixing this.

Hugh

SC-Teuton, 2020-10-23:
> I run Debian under an emulator called Parallels on a MacBook Pro. For a PC running Windows there are several emulators, QEMU and VirtualBox are two of the better known ones.
>
> From your previous [...]

Excellent, thank you, I'll go ahead with that!

Yes, from what I have done, "make sim" was the only command that has failed. The others before worked fine. I am getting the dev kit, so hopefully it shouldn't be a problem, and I have seen Daniel's video on the topic of flashing the files to the watch with the app.

Thank you for your help!

closed 2020-10-26 by daniel-thompson