wasp-os/wasp-os #89: Colmi p8 Gadgetbridge connectivity

cr0mbly, 2020-10-25:
Hi I've finished compiling and flashing the latest build to my Colmi P8 and it works a treat can connect over bluetooth terminal and the watch boots flawlessly. Question around integrating with a mobile device has anyone managed to successfully connect it to GadgetBridge. I'm finding the device isn't picked up there although it is by the android Bluetooth settings (although it can't connect to it.)

Carglglz, 2020-10-25:
Hi @cr0mbly 
>  I'm finding the device isn't picked up there although it is by the android Bluetooth settings (although it can't connect to it.)

See #56 and https://github.com/daniel-thompson/gadgetbridge/commit/732971d7e49d96df9073b121fe7f102f29cbd915

Basically "P8" is not include in "supported names" in Gadgetbridge.

So there are two options:
1) Add "P8" to that list and compile Gadgetbridge yourself.

2) Change the advertised name "P8"  to "Pixl.js"  (I think this one may be easier)

To do this change in `wasp-os/micropython/ports/nrf/boards/p8/mpconfigboard.h`: 
```
#define MICROPY_HW_NUS_NAME         "P8"

```
to

```
#define MICROPY_HW_NUS_NAME         "Pixl.js"

```
And compile. Then GadgetBridge should work as expected. 👍

cr0mbly, 2020-10-26:
Thanks @Carglglz that worked a treat! 

I noticed @daniel-thompson Fork hasn't been pushed upstream to [master](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/devices/banglejs/BangleJSCoordinator.java) is it the intention to do this in the future is there's approval form GadgetBridge or is better to start compiling our own versions of the app to tie into Wasp-OS.

Carglglz, 2020-10-26:
>  is it the intention to do this in the future is there's approval form GadgetBridge

TBH I have no idea, for now the name change does the trick, but if @daniel-thompson fork gets merged into master, it will be nicer I guess. 😄

closed 2020-10-29 by cr0mbly